wpilibsuite/allwpilib #29: PIDController: setSetpoint causes onTarget to be false

virtuald, 2016-05-15:
The core of the problem is that `onTarget` is calculated using the `err` buffer, which gets cleared when a new setpoint is set. This behavior is incorrect -- `onTarget` should not depend on the history of the setpoints (which is how error is calculated), but instead it should only depend on the history of the inputs as that's really the question a user is asking when calling onTarget: "is the input [history] close to where I want it to be?"

To that end, I propose that a buffer of *inputs* is kept in addition to a buffer of errors, and use the inputs to calculate `onTarget` instead. This buffer would only be cleared when the PIDController is disabled.

calcmogul, 2016-05-18:
Actually, we only need to keep a buffer of inputs. Here's some math:

a, b, c, and d are inputs. s is the setpoint. "s - a" is the error with respect to a.

Currently, we average the errors in GetAvgError():
avgError = ((s - a) + (s - b) + (s - c) + (s - d)) / 4
         = (-(a - s) + -(b - s) + -(c - s) + -(d - s)) / 4
         = -((a - s) + (b - s) + (c - s) + (d - s)) / 4
         = -(a + b + c + d - 4 * s) / 4
         = -(a + b + c + d - 4 * s) / 4
         = -(a + b + c + d - 4 * s) / 4
         = (4 * s - (a + b + c + d)) / 4
         = s - (a + b + c + d) / 4

Therefore, GetAvgError() could instead return the current setpoint minus the input average. This would make anything using GetAvgError(), like OnTarget(), more robust to setpoint changes.

PeterMitrano, 2016-05-18:
:+1: for simple math.

virtuald, 2016-11-27:
This was closed in #138, which was reverted... so I presume that this bug still exists? This should be reopened in that case.

calcmogul, 2016-11-29:
How should we go about fixing it this time? Apparently using doubles in a large FIR filter (moving average) causes rounding errors due to so many additions. We could approximate it with an IIR filter and avoid that problem. Using a cutoff frequency isn't necessarily intuitive, but it provides more tuning hints than the size of the FIR filter does. Teams would probably tune both by guessing and checking anyway.

calcmogul, 2017-11-06:
Changing this to an IIR filter breaks the interface. If we're going to do interface breakage, I'd rather move the filter out of PIDController entirely because the state estimation should be separate from the controller. If the user does the following:
```frc::Encoder encoder{1, 2};
frc::LinearDigitalFilter filter = frc::LinearDigitalFilter::SinglePoleIIR(&encoder, 0.3, 0.05);
frc::Spark motor{0};
PIDController pid{kP, kI, kD, &encoder, &motor};
```
the input is filtered instead of the error (which is more correct) and the user can decide what filtering would be best.

calcmogul, 2017-12-04:
Fixed by #38.

PR #138 (2016-08-13): PIDController queue now stores inputs instead of errors
Closes #29.
closed 2017-12-04 by calcmogul