wpilibsuite/allwpilib #30: PIDController: race condition for enable/disable and PIDWrite

virtuald, 2016-05-15:
Note: I have only examined the java implementation, but I assume this holds true for C++ as well.

As a user, if I call `disable()` on a PIDController, I reasonable expect that after that point `PIDWrite` will cease to be called. However, a race condition exists that makes this not the case.

In [calculate()](https://github.com/wpilibsuite/allwpilib/blob/36447d86d85c95ab9cc95b18df95d201ab215513/wpilibj/src/shared/java/edu/wpi/first/wpilibj/PIDController.java#L251), the lock is not held while checking `enabled` -- which may be ok. However, after that line, potentially `enabled` could be changed, and the calculation is performed. PIDWrite is called after the lock is released, but once again enabled could have changed by that point.

One thing I'm torn on is whether the lock should be held while calling PIDWrite. I feel like it should -- but I also feel that could potentially cause deadlocks if the PIDWrite function called something that locked on something that another thread was holding while calling a PID function. Unsure on the best solution there (maybe use different locks, or perhaps atomic primitives instead?).

PeterJohnson, 2016-05-15:
To ensure this, there needs to be a separate recursive mutex that's used only for PIDWrite and Disable.   You do not want to hold the main pidcontroller mutex over the PIDWrite call.  Disable grabs the main pidcontroller mutex and the PIDWrite mutex, then clears the enabled flag.  Calculate grabs in the same order, checks the enable flag, but releases only the pidcontroller mutex before calling PIDWrite, releasing the PIDWrite mutex only after that function has returned.

calcmogul, 2016-07-01:
In addition to this, we could stop the Notifier when Disable() is called. Calculate() doesn't do anything unless the controller is enabled anyway.

commit de63e1c8 (2017-12-05): Fixed race condition between PIDController enable/disable and PIDWrite() call

To make this work in PIDController.java, the use of synchronized had to be
replaced with ReentrantLock and try-catch blocks. The locking in
PIDController.java was made equivalent to PIDController.cpp and some existing
race conditions in PIDController.java were fixed in the process.

Fixes #30.
closed 2017-12-05 by PeterJohnson