wpilibsuite/allwpilib #32: CANTalon: No resource counting to disallow overallocation

ThadHouse, 2016-05-15:
Right now, there is no resource allocation on CANTalon, so you can allocate the same ID with multiple objects. Is this something we would want to fix? It would be a breaking change, at least in java, since we would be throwing a new exception that we were not doing before.

PeterMitrano, 2016-05-15:
sounds super helpful. Do teams struggle with this often? It seems like a stupid error that might be hard to find.

ThadHouse, 2016-05-16:
All of the other classes throw exceptions, so this is the odd class out. And also yes its pretty common especially early in the season. I'll wait until after my HAL proposal to do this, as this can be implemented either at the HAL level or the WPILib level.

erikuhlmann, 2016-09-30:
From a design perspective I think it makes more sense to turn the CANTalon constructor private and create a "factory" static method that initializes singletons for each CAN ID you call it with. (and extend this to the other classes that need a similar pattern)

PeterJohnson, 2016-10-01:
While singletons might have been a cleaner design at inception, making such a change now would break thousands of existing programs.  Per https://github.com/wpilibsuite/allwpilib/blob/master/CONTRIBUTING.md#what-to-contribute, "backwards compatibility is very important for WPILib".

Kevin-OConnor, 2016-10-27:
CANTalon moved to CTRE managed external library

closed 2016-10-27 by Kevin-OConnor