wpilibsuite/allwpilib #33: HAL Changes Proposal

ThadHouse, 2016-05-16:
Currently, the WPILib HAL was made at an arbitrary point in the code. In most places, it is abstracted at a good spot, however there are some places where major improvements could be made in order to make the HAL cleaner, more of an abstraction layer, and remove some current complications from the code. Below is a list of changes that I feel would improve the abstraction layer both for users, and ease of adding a HAL simulator at a later date.

- Change Joysticks to return -1 to 1: The WPILib level should not be manually doing the conversion from binary values to float values. Changing this would enable cleaner simulation and easier updating the scaler value in future years.

- Change PWM values to take -1 to 1: The WPILib should only care about scaler values, and the HAL should be the one converting that to hardware raw output values.

- Move port resource management from the WPILib to the HAL: There is currently limited resource management in the HAL, but most of it is in the WPILib. The WPILib should not need to do this, and moving this to the HAL would clean up the code and enable easier support in future years. In addition, this would allow us to merge MXP resource and regular resource management, instead of those errors actually throwing different types of errors as they currently do.

- Report port constants and DS data counts using the HAL: It seems like this was attempted to do early, but it was never actually completed. Having the constants for the number of ports in 3 places is hard to maintain. 

- Add a Gyro class to the HAL: Currently the WPILib does all the scaling from raw accumulator values to real world values. This is complicated, and seems like the perfect job for a HAL to do. In my opinion, a HAL should always take real world values, and do the internal conversion needed itself.

- Move Encoder and Counter classes to HAL: Currently, counter is OK, but encoders using an encoder when set to 4x, 2 counters when set to 2x, and 1 counter when set to 1x is a complicated abstraction that would be much easier to handle in the HAL. This is currently one of the most complicated class in the WPILib, and one of the most difficult to simulate because you have to check many different things to get the right versions. Plus the raw count scaling changes when different modes are selected.

That’s most of the major changes. There are a few minor ones, mostly involving analog triggers, but I don’t know if it’s possible to do them, and it would be fairly minor anyway.

Anybody have any opinions for or against any of these, or anything else to add?

Kevin-OConnor, 2016-05-16:
**Joysticks** - Yes, while this is not strictly "hardware" that we're abstracting I agree that this makes sense.
**PWM** - Yes, at a minimum it makes sense to move the stuff in SetBounds, SetPosition, SetSpeed, etc. down in to the HAL.
**Resource management** - Yes, this should be done if possible. The barrier in the past has been proper error reporting that provides useful information to teams about what/where the problem is. Some of the changes to the error reporting this year may help with that?
**Port counts** - Yes, the current method is insane. I think this somewhat goes with the resource management one above.
**Gyro** - Maybe, I think this one needs more detail and discussion before an up/down vote.
**Encoders** - I'm not sure on this one. I'll have to dig into this code a little more and think about it to figure out where I stand.

JLLeitschuh, 2016-05-16:
**Resource management** - What about creating static factory methods to create the resources.
So instead of `new Encoder(3);` it would be `Encoder.getForPort(3);` that way we aren't using constructors everywhere and we don't have to deal with the problem of allocating too many of a type.

Kevin-OConnor, 2016-05-16:
That's a pretty significant API change instead of an implementation change...

PeterJohnson, 2016-05-16:
I agree with Kevin... that would be a major breaking API change, and I'm not sure it would provide any real benefit to teams.  What we really want to detect is multiple uses of one port (that's a very common programming error), and that can be done in the HAL based on port number regardless of the way teams instantiate classes at their level (if both the factory function and the constructor raise an error on multiple attempts to access the same port number, what's the benefit of the factory approach?).

PeterJohnson, 2016-05-16:
Thad: Can this change be split up into multiple PRs?  E.g. one for each of the major bullets?  I'm not seeing interdependencies between them, and separate PRs would allow us to integrate piecemeal and separate less certain / more debatable changes like Gyro.

ThadHouse, 2016-05-16:
I agree that we could add the factory approach as a secondary system, but we would have to keep the old constructors, and could probably never depreciate them. It would break way to much code.

Yes my plan was to implement this using separate PRs. Very little of it is interconnected, and could easily be split. In addition, most likely the resource change would be a very large commit, and could potentially be better as 1 PR, but with 4 or so commits.

We can definitely talk about the Gyro and Encoder changes. Both of those changes are to make the HAL more of an abstraction layer. For encoders, having the library determine if the encoder is actually an encoder or 2 counters is kind of hacky, and makes the code fairly complicated and difficult to test. Same thing with Gyros, we are currently sending raw values back from the HAL, and since most other changes involve sending values in real worlds units, the Gyros should too. Technically the encoders could be considered this as well, as the values that get sent through the HAL depend on the encoding type, which shouldn't matter.

PeterMitrano, 2016-05-16:
This seems like a sound plan. Granted, I haven't really spent any time in hal, but I'm going to try to stay involved in this so I know what's going on. I imagine there will be more things I see from the simulation perspective.

bradamiller, 2016-05-16:
I agree with the other comments and that these things would go a long way towards cleaning up the library. Breaking the items up into multiple requests would make it easier to deal with. Once there is better/cleaner separation between the HAL and the higher level library, this would be a good time to add some unit tests to better verify those interfaces. Especially now that the some of the command-based tests are getting refactored into unit tests and run on Travis.

Also, there would have to be the corresponding changes to C++ and Java for this to work - so lots of JNI bindings changes/additions.

Thad, were you just making the proposal or offering to do the work? And if you were doing the work, do you have access to a roboRIO to test changes?

ThadHouse, 2016-05-16:
There would be changes needed at the C++, Java, JNI and HAL level, but I would be willing to try and get as far as I can.

I do not currently have access to a RoboRIO, but I will try to get access to one once I move to California in the middle of June.

virtuald, 2016-05-16:
I have some other annoyances to bring up:

* **Solenoids**: All of them get created when one of them gets created (multiple times, I think?). Would be nice to have a Solenoid be a solenoid
* **Relays**: To create a relay, you call `initializeDigitalPort` and then later on call setRelayXXX. How did it know that it was a relay? Seems like you should have to explicitly create a relay object. 

I would not care about HAL API breakage -- if you mess with that API, then you have to expect to get burned.

ThadHouse, 2016-05-16:
With the resource and pwm value changes, every port was going to have its own port strict down in the HAL with its own initialize and free functions. So the relay thing with get fixed with that. Solenoids will be fixed too, as the current method would not work at all with resource management, and needs to be changed. 

And I figured as much about the API changes. I can think of 5 groups that have actively touched the jni or HAL layers. And 3 of those are us so we should be OK. In the API docs we are looking at we should state that backwards compatibility will only be maintained for the wpilib. The jni and HAL layers are subject to change at any time, at least I think.

ThadHouse, 2016-05-20:
Ok. After some discussion, here is a list of the PRs we will probably be making in the order they will be done in.

- [ ] Rename .hpp files to .h
- [ ] Split Analog files
- [ ] Split Digital files
- [ ] Add support for Handles, but don't add them into the code
- [ ] Add support for Handle based resource management

That can all be done without impacting the WPILib, and should be fairly easy. From that point on we can implement handles everywhere else and do the resource changes.

closed 2016-07-09 by ThadHouse