wpilibsuite/allwpilib #34: WPILib Vision request for a proposal (looking for comments and implementors)

bradamiller, 2016-05-16:
I propose that the vision libraries in WPILIb move from NI Vision to OpenCV for a number of reasons:

- In the real world of college, research, and industry OpenCV is the standard for computer vision. This is what our students will likely eventually be learning
- OpenCV is well supported, well documented, lots of samples, videos and books available for many computer languages.
- While I don't want to mandate this, any team that chooses to use a co-processor will certainly be using OpenCV as their vision library and often NetworkTables for communications to the robot. Although we might not provide a packaged solution for common co-processors, there is enough community activity in this area to make it much easier than it's been in the past.
- GRIP has increased the use of and awareness of OpenCV and made it more approachable.

What problems should we solve:
-  writing OpenCV robot programs should be as easy as using anything else in WPILib. It should work with both C++ and Java for some version of OpenCV (could be version 2 or 3). The libraries, build files, scripts, and anything else required should be packaged with the libraries/plugins.

- Using cameras, both USB and ethernet cameras should just work. The camera settings should be available from the program so that you can, at least, set frame rate, image size and compression, and exposure.

- You should be able to display annotated images on the Driver Station computer, probably through a SmartDashboard plugin. Minimally, the raw would be displayed, but one could use the openCV annotation functions to draw on the images depicting counters, bounding boxes, etc.

- A team just wanting to stream the camera (from the robot) output on their Driver Station, again probably through the SmartDashboard, should be able to do that with minimal effort and minimal performance degradation to the roboRIO robot program.

- These items should run, at least, on the roboRIO. But as a reach goal, bringing back the OpenCV library to run inside of a SmartDashboard plugin was a very attractive option to teams in the past, but we seem to have lost the secret sauce. My understanding is the Ken Streeter and FRC 1591 has resurrected this code and got it working.

- As another reach goal, having Network Tables builds for some of the common co processors would make many peoples lives easier. But not a primary support platform for this team, this is very optional.

- I'd like to see GRIP have the ability to generate a function/class/program or something in one or more languages (C++, Java, Python) that is the few lines of code that implements the pipeline algorithm. In this way, a team could prototype in GRIP, generate code, and stick the code into their robot or dashboard programs.

What do you guys think of these goals?

Brad

PeterJohnson, 2016-05-16:
I'm working on a detailed api and proposal for the camera interface and robot side of dashboard streaming.  Plan to support annotated images as well as camera settings etc.  Image interface to robot code would be OpenCV based.

virtuald, 2016-05-16:
I really like being able to stream images via HTTP, it's a lot more accessible than using a custom semi-documented protocol like it is now.

jkoritzinsky, 2016-07-18:
What's the status of this right now? Any work towards a new Vision API in WPILib?

PeterJohnson, 2016-07-19:
See my pull request above for a draft design doc for the API.  I need to resubmit it against the design-docs repo.  I've not done any implementation work yet however.

PeterJohnson, 2016-11-19:
Current status: #352 has been merged, implementing the final piece of this.  OpenCV is now bundled (instead of nivision) and the OpenCV-based cscore library is used for CameraServer.  There's a few missing pieces (namely Ethernet camera support) but in general this is complete.

SamCarlberg, 2016-12-08:
#388 Adds an API for running an OpenCV pipeline. Generated GRIP code will have the option of implementing the interface so it can slot in seamlessly.

PeterJohnson, 2017-01-08:
This is effectively complete for 2017.

closed 2017-01-08 by PeterJohnson