wpilibsuite/allwpilib #96: CANTalon: no warnings when creating a device that isn't actually plugged in

virtuald, 2016-06-03:
I tried it tonight with a variety of different configurations, in RobotPy and using Java just to double check. When I create a CANTalon object with a device number that isn't connected, I expect the robot to inform me somehow, but there are no exceptions, no error messages, nothing.

I think the CANJaguar used to throw random exceptions at runtime if a Jaguar became disconnected, and I don't think that's a good behavior -- but now the pendulum has swung to the other extreme. I think reasonable expectations are:

At robot startup, if the device doesn't exist:
* Always report an error message to the DS
* Throw an exception if the FMS isn't present

If the device becomes disconnected after the object has been created:
* Always report a warning message to the DS

2423 got bit by this a lot during the season. :(

virtuald, 2016-06-03:
One thing that I didn't try, which I would be interested in knowing, is whether or not the Talon can respond if its communication wires are plugged in properly, but the power has been interrupted. I expect that it would not be able to respond, but if it could respond I would expect it to be treated as disconnected.

calcmogul, 2016-07-02:
Since CAN bus does not provide power, the CANTalon won't be able to send or receive CAN frames while its power source is interrupted. This does not affect other devices on the bus.

Kevin-OConnor, 2016-10-27:
CANTalon moved to CTRE managed external library.

closed 2016-10-27 by Kevin-OConnor