wpilibsuite/allwpilib #151: Provide async notification of DS state changes

PeterJohnson, 2016-07-09:
For some features (like gyro calibration) it would be a nice feature to be able to hook into transitions from disabled to autonomous/teleop.  Currently this can only be done in the RobotBase class, but a feature to monitor these changes in the DS thread and call a callback would be relatively easy to implement and provide this ability more independently of user code.

Starlight220, 2020-05-10:
I would be happy to implement this in Java, but I disagree with a few points:
I think the current DS class would need a lot of added functionality to do this, said functionality already exists in RobotBase. Adding it to the DS would either cause duplicates or a buggy transition. 

> independently of user code

Is there a way to write functional robot code without using `RobotBase`? If not, and the RobotBase class is trivial just like the DS class, then what problem is there with implementing it there? The functionality almost exists.

calcmogul, 2020-08-25:
What kind of stuff would we be running asynchronously? Is the idea that we'd have gyro calibration, for example, happen on the library side on DS transition, and the user wouldn't see it?

Starlight220, 2020-08-25:
I think this would be useful mostly for team libraries, and/or decentralization of the `Robot` class.

calcmogul, 2020-08-25:
You can already get that "decentralization" by making a MySubsystem.teleopInit() function and call that in TimedRobot.telopInit(), and it's much easier to reason about and debug. Team libraries could do the same thing. Are there specific kinds of libraries you had in mind?

Starlight220, 2020-08-25:
not really, it's just a possible reason I thought of. If this isn't a useful feature anymore, close this issue.

closed 2020-09-14 by PeterJohnson