wpilibsuite/allwpilib #206: Make SendableChooser dashboard widget have priority over robot

bradamiller, 2016-08-18:
When a team selects an autonomous mode on the dashboard, then restarts (or starts) the robot, the robots default value for the sendable chooser overwrites the user selected value since it came in after the dashboard started. The desktop value should have priority so they can set preferences before the robot is finished booting. This seems to make sense unless someone can come up with a counter-example where the newer robot value should take priority over the users input at the dashboard.

I think the list of values is showing up because the dashboard layout is saved.

AustinShalit, 2016-08-18:
Could we flag the selected value as persistent to achieve the same result?

ThadHouse, 2016-08-18:
Persistent values wouldn't work for this in certain cases. For instance, if the persistent value is set to mode 2, but the SmartDashboard selects mode 3 for the next match, mode 2 would be the one selected. This is because when a client connects to a server it overwrites any client entries with ones the server has.

One way to solve this would be to use SetDefault on the RobRio side to set the value at startup, and then a change in the NetworkTables protocol to allow certain values to be flagged as `client override`, where if the server is sending a value on initial connection that exists on the client already, the client ignores it, and sends the proper value back to the server.

@PeterJohnson what do you think?

bradamiller, 2016-08-18:
I was thinking that if the user explicitly set the value in the UI in the dashboard, they meant it to have that value and it shouldn't get overridden by the default value that is in the robot on startup.

AustinShalit, 2016-09-01:
Closes with wpilibsuite/SmartDashboard#6

PeterJohnson, 2016-09-05:
wpilibsuite/SmartDashboard#6 has been merged.

closed 2016-09-05 by PeterJohnson