wpilibsuite/allwpilib #229: Check and see if 1x and 2x encoders can use FPGA counter object

ThadHouse, 2016-09-13:
It might be the case where a user wants more then 8 1x or 2x encoders, however since by default since those always use counters, you can't have more then 8. With the move of encoders to the HAL, it should be possible to allocate 1x or 2x encoders to FPGA Encoder objects fairly easily, as long as the FPGA Encoder object can handle 1x or 2x decoding. Worth checking out. If that doesn't work, it might be worth printing an error message stating try switching to a 4x encoder, as those might be available.

ThadHouse, 2017-10-15:
After some thought, automatically switching encoder decoding type can lead to unexpected bugs in existing code, so not a worthwhile breakage. I can implement a message telling users they can switch to a 4x encoder if that seems helpful.

Kevin-OConnor, 2024-06-20:
There were 3 teams that used 8 encoders of any type in 2024 so I think this can be closed as a "Won't fix"

closed 2024-06-20 by Kevin-OConnor