wpilibsuite/allwpilib #233: DriverStation::IsNewControlData is incorrect

virtuald, 2016-09-14:
Current C++ implementation is:

```
bool DriverStation::IsNewControlData() const {
    return m_newControlData.tryTake() == false;
}
```

This is correct IFF you are calling it a lot and never forget to call it. But really, it's just a bad way of doing this. [The Java implementation is actually correct](https://github.com/wpilibsuite/allwpilib/blob/master/wpilibj/src/athena/java/edu/wpi/first/wpilibj/DriverStation.java#L533).

ThadHouse, 2016-09-14:
I believe #187 fixes this issue. I realized that there were some other issues as well with using a semaphore, so that PR uses an atomic bool in C++ and Java.

PeterJohnson, 2016-10-27:
Closed with #187.

closed 2016-10-27 by PeterJohnson