wpilibsuite/allwpilib #284: AnalogInput::GetSampleRate Segmentation Fault

JacisNonsense, 2016-10-21:
It would seem that in the C++ version of the WPILib library, any call to `AnalogInput::GetSampleRate()` (wpilibc athena, 2017 Beta 1)  causes a segmentation fault. It seems that this is due to the `HAL_GetAnalogSampleRate(int32_t *)` function (hal athena, 2017 Beta 1), specifically at line 139.

```
138 | double HAL_GetAnalogSampleRate(int32_t* status) {
139 |  uint32_t ticksPerConversion = analogInputSystem->readLoopTiming(status);
140 |  uint32_t ticksPerSample =
141 |      ticksPerConversion * getAnalogNumActiveChannels(status);
142 | return static_cast<double>(kTimebase) / static_cast<double>(ticksPerSample);
143 | }
```

The issue appears to be in the call to `analogInputSystem->readLoopTiming(status)`, which is interesting as the backtrace does not enter the `readLoopTiming` call, stopping at `HAL_GetAnalogSampleRate`. Running `gdb` confirms this claim.

 My initial assumption would be that this issue is due to the linkage to the ChipObject library. Please note that this is the 2017 Beta 1 library running on a 2016 RoboRIO image. 

To dispel the obvious: This call is made after `RobotBase->StartCompetition` (called in periodic functions), and is also made only after at least 1 `AnalogInput` instance has been created.

Regards,
~Jaci

ThadHouse, 2016-10-23:
I cannot seem to reproduce this with the 2017 image and the latest mater of the repo, so this was probably just a linkage issue. I'll keep this issue open until we release beta 2 with the 2017 image and can check this again.

However, this did bring to my attention where `AnalogInput::GetSampleRate()` can segfault if an analog input has not been created before that method is called. I've got some changes to make, but PR #285 should fix that issue.

JacisNonsense, 2016-10-23:
No problem. Is the 2017 image currently available? I'll try and confirm if it's an image issue in that case.

333fred, 2017-05-07:
Closing as there have been no updates. @JacisNonsense, if this is still an issue, please tag me and I'll reopen.

closed 2017-05-07 by 333fred