wpilibsuite/allwpilib #304: WPILib Maven Publishing and Cache Invalidation

JacisNonsense, 2016-10-27:
An issue regarding WPILib's maven publishing has recently come to my attention through [this peculiar issue](https://github.com/Open-RIO/GradleRIO/issues/15#issuecomment-256708019) on GradleRIO.

In short, versions are being incorrectly published to maven. All new releases are being published under the same version `0.1.0-SNAPSHOT`, with subversions for each release (and beta release). Because of this, anything that fetches from maven (gradle being a prime example) will only fetch once for its cache, but this cache will not invalidate with new releases as it sees no new versions published, and so the build version of a library can entirely depend on the system it's running on, breaking not only community forks and projects, but also can lead to an ambiguity in "it works on my machine", however for other members as well as CI services, the build breaks.  

Compare this to a maven repository that is properly versioned, such as [GradleRIO](http://dev.imjac.in/maven/jaci/openrio/gradle/GradleRIO/) or even the [development branch of WPI's repository](http://first.wpi.edu/FRC/roborio/maven/development/edu/wpi/first/wpilibj/wpilibJavaFinal/), you can see how this becomes an issue.

333fred, 2016-10-27:
This was recently fixed. Going forward, all of our dependencies will have distinct version numbers. This is available in the development maven repo now, and will start appearing in the release repo as we start publishing betas.

333fred, 2016-10-27:
Note that there are new maven dependency names for some of our dependencies. I'll be updating the docs on ScreenSteps soon.

JacisNonsense, 2016-10-27:
Sounds good. Is there a chance we can get the latest, 2016 (i.e. public) stable release for WPILibJ pushed to the release tree? This will fix issues for everyone on GradleRIO running offseason / training courses and projects.

333fred, 2016-10-27:
If you just set your version to 0.1.0-SNAPSHOT and invalidate the cache once, it should grab the latest version. I'm not going to have time to do this in the next few days, so that might be your best option.

JacisNonsense, 2016-10-27:
This sets the fetched version to `2017.1.0-beta-1`, which doesn't appear to be entirely stable on the 2016 RoboRIO image, and also removes all reference to manufacturer devices like CANTalon, breaking a lot of existing codebases. 

For now I'll note it down as a dependency bug and try and work on a temporary GradleRIO hotfix in the morning.

333fred, 2016-10-27:
Setting the version to 0.1.0-SNAPSHOT fetches beta 1? That shouldn't be. I know GradleRIO uses +, are you sure that you're setting the version to 0.1.0-SNAPSHOT?

JacisNonsense, 2016-10-27:
Doing a cache invalidation and refetching does indeed fetch from [0.1.0-SNAPSHOT](http://first.wpi.edu/FRC/roborio/maven/release/edu/wpi/first/wpilibj/wpilibJavaFinal/0.1.0-SNAPSHOT/), evident from both Gradle's log, cache location and the listing in WPI's maven. If you look at the [repo](http://first.wpi.edu/FRC/roborio/maven/release/edu/wpi/first/wpilibj/wpilibJavaFinal/0.1.0-SNAPSHOT/), releases are fairly standard for 2016 (Jan & Feb), however new snapshots are being posted for 2017 Beta Testing, from 13 Oct. `0.1.0-SNAPSHOT` is the only version listed in the release tree.

333fred, 2016-10-27:
I can try to find some time to do this. Can you give me a list of the dependencies that you'd need published?

JacisNonsense, 2016-10-27:
The stable 2016 releases of NetworkTables (arm / desktop) and WPILibJ (both w/ sources) should be enough to satisfy most projects. I believe the most recent for WPILibJ is 2016 Release 5

333fred, 2016-10-27:
Alright, I'll try to get to this in the next few days.

keplersj, 2016-10-30:
Can confirm. Getting compilation errors in a 2016 project using GradleRIO due to the removed CANTalon class.

keplersj, 2016-10-30:
/cc @jmchs-robotics/mentors @jmchs-robotics/2017-software

333fred, 2016-10-30:
ntcore will soon be up under version 3.0.0, and wpilibc/j will be up under version 2016.5.1. Should be up sometime within the next five minutes, whenever the sync job runs next.

closed 2016-10-30 by 333fred

333fred, 2016-10-30:
A note about ntcore: I've published the `linux` and `arm` classifiers, but not desktop. It would be a bit of a pain to get the full `desktop` classifier, as I don't have easy access to a Mac machine without setting up a full Jenkins job just to orchestrate the build. So, if you need to develop on desktop, use the latest version. It's fully compatible with the 2016 version, just some new features that were added.

JacisNonsense, 2016-10-30:
No problem. NT-Core seems to be stable for the moment, and the only projects (at least that I know of) that require it fully are Toast projects, which are on a stable 2016 build currently. 

Thanks for setting that up, this should resolve numerous issues

JacisNonsense, 2016-10-30:
@keplersj please run `./gradlew build --refresh-dependencies` to refresh Gradle's cache and build your project. The build might take a minute or two depending on your internet speed, but from there on our just `./gradlew build` like normal should work just fine

keplersj, 2016-10-30:
@JacisNonsense That did it! Thanks!