wpilibsuite/allwpilib #327: Improve hardware test platform

PeterJohnson, 2016-11-04:
The current hardware test platform (used by Jenkins) is a little finicky because it uses mechanical components, and also does not cover several areas of the hardware such as I2C.  Using a component such as an Arduino to measure PWM signal characteristics, drive analog inputs, interface to SPI and I2C, etc, may be one option to eliminating mechanical components, improving coverage, and making it easier for developers to replicate the test platform at their own locations.

virtuald, 2016-11-29:
this would be a great thing to make available to teams who wanted to do their own hardware testing -- I feel like I saw someone mentioning something like this on CD.

calcmogul, 2026-08-24:
The integration test platform is no longer used because the integration tests that ran on it were removed in #7692.

closed 2026-08-24 by calcmogul