wpilibsuite/allwpilib #342: Uundocumented exception in an undocumented native method

nimrod46, 2016-11-11:
The method "getImage" in the axisCamera class uses an native method called: `_Priv_ReadJPEGString_C(image.getAddress(), getByteBufferAddress(string_buf), stringLength);`
Aftter several calls that are made to this method I get an imaqError -123673104 which is an undocumented exception, this exception is probably because the java GC does not free the image allocated by the API.

This exception has been discussed here: https://www.chiefdelphi.com/forums/showthread.php?t=152212

Kevin-OConnor, 2016-11-11:
getImage does not describe a unique method in AxisCamera. Can you please post a minimal sample that you are seeing an issue with?

nimrod46, 2016-11-11:
@Kevin-OConnor 
` public HSLImage getImage() throws NIVisionException {
    HSLImage image = new HSLImage();
    this.getImage(image);
    return image;
  }` 
lines 138 - 142;

Kevin-OConnor, 2016-11-11:
That tells me which getImage but still does not tell me what you are doing with it. Can you please post a minimal sample that you are seeing an issue with?

nimrod46, 2016-11-11:
@Kevin-OConnor Ohh, sorry.
Here you go:

`public newParticleAnalysisReport imageProcessing() {

		
		try {
			HSLImage inputImage = camera.getImage();
			Image processImage = NIVision.imaqCreateImage(NIVision.ImageType.IMAGE_U8, 0);

			/* Ranges for the HSV variables in the Threshold operation */
			NIVision.Range hueRange = new NIVision.Range(HUE_LOW, HUE_HIGH);
			NIVision.Range saturationRange = new NIVision.Range(SATURATION_LOW, SATURATION_HIGH);
			NIVision.Range valueRange = new NIVision.Range(VALUE_LOW, VALUE_HIGH);

			try {
				NIVision.imaqColorThreshold(processImage, inputImage.image, 255, NIVision.ColorMode.HSV, hueRange,
						saturationRange, valueRange);

			} catch (Exception e) {
				e.printStackTrace();
			}

			/*
			 * First filter operation of the Threshold image - "Particle Filter"
			 */
			NIVision.ParticleFilterCriteria2[] criteria = new ParticleFilterCriteria2[2];
			criteria[0] = new ParticleFilterCriteria2(NIVision.MeasurementType.MT_AREA, MIN_AREA, MAX_AREA, 0, 0);
			criteria[1] = new ParticleFilterCriteria2(NIVision.MeasurementType.MT_RATIO_OF_EQUIVALENT_RECT_SIDES, 9, 15, 0, 0);

			ROI roi = NIVision.imaqCreateROI();
			ParticleFilterOptions options = new ParticleFilterOptions(0, 0, 1);	
			NIVision.imaqParticleFilter3(processImage, processImage, criteria[0], 1, options, roi);
			NIVision.imaqParticleFilter3(processImage, processImage, criteria[1], 1, options, roi);

			newBinaryImage binaryImage = new newBinaryImage();
			binaryImage.image = processImage;

			/*
			 * Second filter operation of the Threshold image -
			 * "Remove Small Object"(manual)
			 */
			/* Sending a Particle Report of the largest object remaining */
			if (counter >= 2) {
				System.out.println("number of particales: " + binaryImage.getNumberParticles());
				counter = 0;
			}
			counter++;
			if (binaryImage.getNumberParticles() > 0) {
				particleArray = binaryImage.getOrderedParticleAnalysisReports(binaryImage.getNumberParticles());
				maxArea = particleArray[0].particleArea;
				particleReport = particleArray[0];
				for (newParticleAnalysisReport par : particleArray) {
					if (maxArea < par.particleArea) {
						maxArea = par.particleArea;
						particleReport = par;
					}
				}
			}	
		} catch (Exception e) {
			e.printStackTrace();
		}

		return particleReport;
	}`

Kevin-OConnor, 2016-11-16:
You are definitely running out of memory. Even if the destructor called free(), by the time the GC got around to actually freeing this memory you would likely have an issue anyway.

Either call free() on all of the images you are allocating or create one of each type as a static or in a scope where they won't go out of scope every time and re-use them (what I would recommend).

For 2017 this code is being removed from WPILib (but will still be available as an external module). The code is moving to this repo: https://github.com/wpilibsuite/nivision

closed 2016-11-16 by Kevin-OConnor