wpilibsuite/allwpilib #374: protobuf sources

ariovistus, 2016-11-25:
In https://github.com/wpilibsuite/allwpilib/tree/master/simulation/JavaGazebo/src/main/java/gazebo/msgs, 
there is protocol buffers generated code. It appears some of the files are generated from 
https://github.com/wpilibsuite/allwpilib/tree/master/simulation/gz_msgs/src/main/proto,
but the rest I couldn't find [in my 10 seconds of searching].

Anyone know where the sources are?

calcmogul, 2016-11-25:
Some of the sources aren't included, but I was able to remake them based on comments in the generated output. See the latest commit on https://github.com/calcmogul/allwpilib/commits/cleanup-sim.

ariovistus, 2016-11-26:
ya, I notice the source was embedded a while after I posted, but I figured someone will have already put it back together :) 
thank you thank you

closed 2016-11-26 by ariovistus

virtuald, 2016-11-26:
The last time I messed with this, I believe that the missing sources came directly from the gazebo source code.

ariovistus, 2016-11-26:
hey, it is:
https://bitbucket.org/osrf/gazebo/src/5ef70d513ac17fb393d5fe95da44b49c2a5cbade/gazebo/msgs/?at=default

dustin is better than google search

@calcmogul there are some minor issues with your protobuf sources that prevent compilation, fyi