wpilibsuite/allwpilib #413: Dependencies for WPILibC on Maven

JacisNonsense, 2016-12-20:
Currently, WPILIbC is being published to the WPILib Maven. 
Unfortunately, this maven entry does not include any dependencies in the maven pom, nor any packaged with the zip. Given that WPILibC depends on NTCore, CSCore, WPIUtil and the HAL, While the HAL follows the same versioning scheme as WPILibC, NTCore, CSCore and WPIUtil are all separate projects with their own versions for each release of WPILibC. 

The issue is that if a project is fetching a prior version of WPILibC from Maven, it can become challenging to determine what the correct version of the other libraries should be, as the latest available may have compatibility issues with prior versions of WPILibC.

ThadHouse, 2016-12-20:
Note Java will have this issue as well, since it no longer includes ntcore, cscore or jni in the jar itself. I would much rather prefer figuring out how to put the dependency version in the pom. I'm hoping we can do that fairly easily using things built into gradle, as I think it's too late before the season to add another package structure to our maven repo.

JacisNonsense, 2016-12-21:
I believe this can be done in Gradle. I'm not sure if it's the same with the version of the Maven Plugin that is used in this repo, but you can modify the Pom directly from the DSL https://github.com/Open-RIO/ToastAPI/blob/development/build.gradle#L228-L240

dixonwille, 2017-01-08:
@ThadHouse Hey having an issue with the code running on the RIO where I think your comment comes into play. I am using gradle to grab dependencies from the maven repository. I have opencv, cscore, athena, athena-jni, NetworkTables all being included in the jar. Yet when going into the rio an try and run robotCommand, Java complains it cannot find JNI. Not sure what is missing...

```
Exception in thread "main" java.lang.UnsatisfiedLinkError: no wpilibJavaJNI in java.library.path
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1857)
	at java.lang.Runtime.loadLibrary0(Runtime.java:870)
	at java.lang.System.loadLibrary(System.java:1119)
	at edu.wpi.first.wpilibj.hal.JNIWrapper.<clinit>(JNIWrapper.java:22)
	at edu.wpi.first.wpilibj.RobotBase.initializeHardwareConfiguration(RobotBase.java:171)
	at edu.wpi.first.wpilibj.RobotBase.main(RobotBase.java:194)
```

ThadHouse, 2017-01-08:
No native libraries are included anymore in any of the WPILib Java Jars. In order to run them, you will need to deploy all the files inside the following following dependencies

* athena-jni
* athena-runtime
* opencv-jni

In addition, you will need to get the athena-uberzip artifact from the cscore repo, and deploy the libcscore.so file, and all the files ending in .3.1. You do not need to extract the rest of the files from that archive.

ThadHouse, 2017-01-08:
Oh and they all need to be deployed to `/usr/local/frc/lib`. Along with running `ldconfig` after deploying the libraries.

dixonwille, 2017-01-09:
Thanks for the help!

333fred, 2017-05-07:
I believe the cause of the issue here (for Java) is that since we're publishing zip files and not the actual compilation output, Maven doesn't have the correct dependencies. The C++ root cause is that Maven was in no way designed to do what we're doing with it. 😄 @ThadHouse, we'll want to fix this as part of our artifact cleanup.

jrmihalick, 2018-01-15:
@ThadHouse After updating the firmware on our RoboRIO from the 2017 firmware to the 2018 firmware, we are getting the same error that @dixonwille mentioned on Jan 8, 2017.  That is, we are getting this error:

`Exception in thread "main" java.lang.UnsatisfiedLinkError: no wpilibJavaJNI in java.library.path
    at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1867)
    at java.lang.Runtime.loadLibrary0(Runtime.java:870)
    at java.lang.System.loadLibrary(System.java:1122)
    at edu.wpi.first.wpilibj.hal.JNIWrapper.<clinit>(JNIWrapper.java:22)
    at edu.wpi.first.wpilibj.RobotBase.initializeHardwareConfiguration(RobotBase.java:171)
    at edu.wpi.first.wpilibj.RobotBase.main(RobotBase.java:194)`

Sure enough, when we check the /usr/local/frc/lib directory on the RIO, there is no wpilibJavaJNI.so file there (or anything similarly named).  Can anyone suggest why this file might now be missing OR where we can get the file?  I have looked in MANY of the artifacts on the maven repository at `http://first.wpi.edu/FRC/roborio/maven/release` and I am not finding the file in any of those artifacts.  Your help would be greatly appreciated.

ThadHouse, 2018-01-15:
Are you guys using the 2018 eclipse or gradlerio release? The name of that library changed for 2018, and so did more of the maven locations.

jrmihalick, 2018-01-15:
We are deploying with 2018 eclipse and just started to learn about gradlerio, but haven't gotten very far with Gradlerio.  We'd be happy to start using gradlerio if that will fix this problem.

ThadHouse, 2018-01-15:
Hmm. 2018 eclipse should work just fine. You definitely upgraded the plugins to 2018, right? And did an Rio reimage?

jrmihalick, 2018-01-15:
@ThadHouse Thanks, your questions helped us figure out what the issue was.  As I mentioned, we had new firmware deployed to the robot, but we had the remnants of a working gradle build that was working for 2017, but we didn't have fully updated to work with 2018 and GradleRIO.  We got the project updated to use GradleRIO and we now at least have a different error that we can work on tomorrow.  We no longer are getting the error I mentioned above.  Thanks again!

jrmihalick, 2018-01-15:
Bottom line is that we were running code with 2017 WPI lib dependencies included, but deploying to a RIO with 2018 firmware/libraries installed on it.  Thus the reason our robot code was trying to load a library which no longer exists.

ThadHouse, 2018-05-13:
This hasn't been 100% fixed, but with the merge to the uber repo, all utilities now have the same version. So for users, this should be much easier. I'm going to leave the issue open though.

PeterJohnson, 2019-08-26:
This should no longer be an issue with the move to GradleRIO.

closed 2019-08-26 by PeterJohnson