wpilibsuite/allwpilib #427: C++ SmartDashboard::init() isn't called until StartCompetition

PeterJohnson, 2016-12-30:
While other globals are initialized prior to user class construction, SmartDashboard is not, so if the user class (or any of its instance variables) try to call a SmartDashboard function from within their constructors (rather than e.g. from RobotInit() or later), a crash will result.

The fix should be straightforward: move the SmartDashboard::init() call to either START_ROBOT_CLASS or the RobotBase constructor.

bradamiller, 2016-12-30:
Does this solution still work if someone creates a static (global) instance of an object? Those constructors run before any other code is run, and in random order.

PeterJohnson, 2016-12-31:
No.  A static global instance will run even before `HAL_initialize` gets called.  There's really not much we can do about that case, other than to document "don't do that", unless we want to add initialization checks everywhere that pessimize performance for everyone just to fix that corner case.

PR #428 (2017-01-01): Call SmartDashboard::init() in RobotBase constructor.
Fixes #427
commit 65b091a3 (2017-01-01): Call SmartDashboard::init() in RobotBase constructor. (#428)

Fixes #427
closed 2017-01-01 by PeterJohnson