wpilibsuite/allwpilib #429: Java Threads do not respond to SIGTERM

AustinShalit, 2016-12-30:
If a team creates a program like this:
```
public class Robot extends IterativeRobot {
    public void robotInit() {
        Thread t = new Thread(() -> {
            while (true) {
            }
        });
        t.start();
    }
}
```
The program will fail to die when sent a SIGTERM.

We should document that teams should **_not_** do this and instead do this:
```
public class Robot extends IterativeRobot {
    public void robotInit() {
        Thread t = new Thread(() -> {
            while (!Thread.interrupted()) {
            }
        });
        t.start();
    }
}
```

We should probably also ask NI to update `frcKillRobot.sh` to send a SIGKILL if the user program does not die after a few tenths of seconds.

nightpool, 2017-01-01:
Why is daemonizing the threads "worse"?

AustinShalit, 2017-01-01:
I think that a best case is you do a combination of the two (check the interrupted flag and set as daemon).

Just doing the daemon thread is worse because if in the future a team wants to stop processing images they would need to think of a way to do that.  They might do [Thread.stop which is really bad](https://docs.oracle.com/javase/8/docs/api/java/lang/Thread.html#stop--).  If they check the interrupted flag all they need to do is call `Thread.interrupt` to stop the thread.

nightpool, 2017-01-01:
Sure, but we can't predict whether they'll want to stop the thread in user
code or not. They may want to, but it adds a large layer of complexity that
I don't think it's worth worrying about in documentation, and the solution
is very obvious to a team that goes looking for it. (and is nonobvious for
a team that isn't reading the thread documentation—how are they going to
find the interrupted method anyway?)

Furthermore, Thread.stop is signposted very well in the documentation,
marked as deprecated, and furthermore isn't very unsafe when it comes to
the given usecase (aborting a thread that's just consuming images)

AustinShalit, 2017-01-10:
Just for reference: https://www.chiefdelphi.com/forums/showthread.php?threadid=153362

nightpool, 2017-01-10:
@AustinShalit I'm not sure what that thread has to do with the difference
between setDaemon/setInterrupted. It seems like either would have worked
for that user, correct?