wpilibsuite/allwpilib #448: Packaging of 3rd Party Libraries

JacisNonsense, 2017-01-08:
Apologies for posting this here, as this is a bug in 3rd party libraries (specifically, CTRE's toolsuite), however I believe this should be put out to all 3rd parties delivering libraries.

Currently, some distributors (CTRE in this case), are bundling way too much into their `.jar` files. Below is a screenshot of a decompilation of CTRE's Toolsuite Jar.
[image]

For reasons unknown to me, they appear to be including eclipse setup, `WPILib`, `NetworkTables` and their own library (???) in the library, which all together account for about 98% of the jar file's final size. The inclusion of these is unnecessary and inflates user jars by an insane margin.

333fred, 2017-01-08:
Tagging @ozrien

ThadHouse, 2017-01-08:
For new vendors, we are heavily recommending our [vendor template](https://github.com/wpilibsuite/vendor-template) in order to help avoid issues like that, and make sure all the the zips and jars are properly formatted. However I believe CTRE is using a custom build setup that doesn't have safeties to ensure that doesn't happen.

ozrien, 2017-01-08:
Tagging @JCaporuscio 
I don't think that was deliberate, most likely just an oversight.  We'll fix it in the next release.
For the time being is that preventing you from using the API?

JacisNonsense, 2017-01-10:
It's not preventing usage at the moment

JCaporuscio, 2017-01-28:
This has been fixed as of release 4.4.1.9 of the CTRE Toolsuite Installer.

closed 2017-01-28 by 333fred