wpilibsuite/allwpilib #452: Simulation library for 2017 references outdated libboost shared object

mrosen, 2017-01-09:
The pre-built Gazebo plugins for 2017, here:  http://first.wpi.edu/FRC/roborio/maven/release/edu/wpi/first/wpilib/simulation/simulation/2017.1.1/

references libboost1.54.0

This conflicts with the newer versions of the libraries (already installed) and prevents Gazebo from loading the plugins:

[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 10.0.0.106
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 10.0.0.106
[Wrn] [msgs.cc:1655] Conversion of sensor type[sonar] not suppported.
[Err] [Plugin.hh:165] Failed to load plugin /home/msr/wpilib/simulation/plugins/libclock.so: libboost_system.so.1.54.0: cannot open shared object file: No such file or directory
[Err] [Plugin.hh:165] Failed to load plugin /home/msr/wpilib/simulation/plugins/libdc_motor.so: libboost_system.so.1.54.0: cannot open shared object file: No such file or directory

PeterMitrano, 2017-01-10:
AFAIK no one has been maintain the simulation builds. I suspect most people now want ubuntu 16.04 and gazebo7. I think it makes sense to switch (and therefore update boost and everything else). Should be just a matter of doing the upgrades on our build machines and updating the release

PeterJohnson, 2018-07-27:
With the Gazebo changes for 2019, this should no longer be applicable.

closed 2018-07-27 by PeterJohnson