wpilibsuite/allwpilib #453: Build error on OSX 10.11.6

thefangbear, 2017-01-13:
`./gradlew build` produces the following output:
```
:downloadArmWpiutil UP-TO-DATE
:downloadWpiutil UP-TO-DATE
:hal:compileHALAthenaSharedLibraryHALAthenaCpp
allwpilib/hal/lib/athena/AnalogGyro.cpp:10:18: fatal error: chrono: No such file or directory
 #include <chrono>
                  ^
compilation terminated.

In file included from allwpilib/hal/lib/athena/AnalogAccumulator.cpp:10:0:
allwpilib/hal/lib/athena/AnalogInternal.h:12:18: fatal error: memory: No such file or directory
 #include <memory>
                  ^
compilation terminated.

allwpilib/hal/lib/athena/Accelerometer.cpp:12:19: fatal error: cassert: No such file or directory
 #include <cassert>
                   ^
compilation terminated.

allwpilib/hal/lib/athena/AnalogInput.cpp:10:17: fatal error: mutex: No such file or directory
 #include <mutex>
                 ^
compilation terminated.

In file included from 
allwpilib/hal/lib/athena/AnalogInternal.cpp:8:0:
allwpilib/hal/lib/athena/AnalogInternal.h:12:18: fatal error: memory: No such file or directory
 #include <memory>
                  ^
compilation terminated.

In file included from 
allwpilib/hal/lib/athena/Compressor.cpp:12:0:
allwpilib/hal/lib/athena/PCMInternal.h:12:18: fatal error: memory: No such file or directory
 #include <memory>
                  ^
compilation terminated.

In file included from 
allwpilib/hal/lib/athena/AnalogOutput.cpp:10:0:
allwpilib/hal/lib/athena/AnalogInternal.h:12:18: fatal error: memory: No such file or directory
 #include <memory>
                  ^
compilation terminated.

In file included from 
allwpilib/hal/lib/athena/AnalogTrigger.cpp:10:0:
allwpilib/hal/lib/athena/AnalogInternal.h:12:18: fatal error: memory: No such file or directory
 #include <memory>
                  ^
compilation terminated.

In file included from 
allwpilib/hal/lib/athena/cpp/priority_mutex.cpp:8:0:
allwpilib/hal/include/HAL/cpp/priority_mutex.h:11:17: fatal error: mutex: No such file or directory
 #include <mutex>
                 ^
compilation terminated.

In file included from 
allwpilib/hal/lib/athena/Counter.cpp:11:0:
allwpilib/hal/lib/athena/DigitalInternal.h:12:18: fatal error: memory: No such file or directory
 #include <memory>
                  ^
compilation terminated.

...output for 27 more failed operation(s) continued in file:///Users/derros/Projects/lsrobotics/chinook/allwpilib/hal/build/tmp/compileHALAthenaSharedLibraryHALAthenaCpp/output.txt.
:hal:compileHALAthenaSharedLibraryHALAthenaCpp FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':hal:compileHALAthenaSharedLibraryHALAthenaCpp'.
> Multiple build operations failed.
      C++ compiler failed while compiling AnalogGyro.cpp.
      C++ compiler failed while compiling AnalogAccumulator.cpp.
      C++ compiler failed while compiling Accelerometer.cpp.
      C++ compiler failed while compiling AnalogInput.cpp.
      C++ compiler failed while compiling AnalogInternal.cpp.
      C++ compiler failed while compiling Compressor.cpp.
      C++ compiler failed while compiling AnalogOutput.cpp.
      C++ compiler failed while compiling AnalogTrigger.cpp.
      C++ compiler failed while compiling priority_mutex.cpp.
      C++ compiler failed while compiling Counter.cpp.
      ...and 27 more failures.
  See the complete log at: file://xxxxxxxxxxxxxx/allwpilib/hal/build/tmp/compileHALAthenaSharedLibraryHALAthenaCpp/output.txt

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output.

BUILD FAILED

Total time: 3.01 secs

```

* Additionally, when using `arm-frc-linux-gnueabi-gcc` to compile a random test file, it outputs something really wierd:
```
arm-frc-linux-gnueabi-gcc test.c
clang: error: unknown argument: '-meabi=5'

```

Consulting Google produced little help.
Shouldn't the FRC-provided cross-gcc have no connection with clang whatsoever?

PeterJohnson, 2017-07-08:
I can't reproduce this on Sierra (OSX 10.12.5).  It definitely seems like something isn't set up correctly in your local environment; clang should definitely not be getting used as part of the cross-gcc.  What does `arm-frc-linux-gnueabi-gcc --version` print?

PeterJohnson, 2017-11-20:
Without more information, I'm closing this.  Feel free to reopen if you're still having an issue.

closed 2017-11-20 by PeterJohnson

garrett-davidson, 2018-01-15:
I just spent the last two days trying to track down this same issue. Eventually, I was able to discover that my gcc install via Homebrew messed with something in the FRC toolchain installer. Once I uninstalled gcc in Homebrew and reinstalled the toolchain, everything started working again.

ShadeAJ1, 2020-01-23:
Doesn't work for me. I get this error
`clang: error: unknown argument '-meabi=5', did you mean '-mabi=5'?
make: *** [bin/main.cpp.o] Error 1
`

ThadHouse, 2020-01-23:
Are you trying to build on 10.11? I highly doubt that would work anymore, this issue is a few years old. 

On newer os's the roboRIO compiler can be installed with ./gradlew installRoboRioToolchain 

This will work on the latest master. Likely requires 10.14 at minimum, we haven't tested with anything earlier.

disposedtrolley, 2020-03-09:
> Doesn't work for me. I get this error
> `clang: error: unknown argument '-meabi=5', did you mean '-mabi=5'? make: *** [bin/main.cpp.o] Error 1 `

Not sure if you're still running into this issue, but `gcc` on macOS is aliased to `clang` by default, so the script is trying to run `clang` with the `meabi=5` flag which it doesn't support.

You can confirm this by running `gcc --version` and seeing if the output includes something like `Apple clang version 11.0.0 (clang-1100.0.33.17)`.

Have a look here for more info https://apple.stackexchange.com/questions/245891/installed-gcc-with-homebrew-now-how-to-use-that-gcc-instead-of-clang

MagicalGirlXi, 2020-03-21:
> Not sure if you're still running into this issue, but `gcc` on macOS is aliased to `clang` by default, so the script is trying to run `clang` with the `meabi=5` flag which it doesn't support.
>
> You [...]

After add alias gcc="gcc-9" , the error still exist. I use "gcc-arm-none-eabi-8-2019-q3-update" version.