wpilibsuite/allwpilib #456: Hi,

andrasfe, 2017-01-15:


andrasfe, 2017-01-15:
I've been trying to install the whole FRC suite while trying to make simulation work. On several operating systems, including Ubuntu 16.04, 15.05 and MAC Sierra as well as El Capitan, eventually ended up having the same issue:
For java simulation of GearBots, looks like compilation is succesfull, however, I get the following runtime error:

    [java] java.lang.NoClassDefFoundError: edu/wpi/first/wpilibj/command/TimedCommand
     [java] Initialized simulator/dio/1/2
     [java] 	at java.lang.ClassLoader.defineClass1(Native Method)Initialized simulator/dio/3/4
     [java] 	at java.lang.ClassLoader.defineClass(ClassLoader.java:763)
     [java] 	at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:142)
     [java] 	at java.net.URLClassLoader.defineClass(URLClassLoader.java:467)
     [java] 	at java.net.URLClassLoader.access$100(URLClassLoader.java:73)
     [java] 	at java.net.URLClassLoader$1.run(URLClassLoader.java:368)
     [java] 	at java.net.URLClassLoader$1.run(URLClassLoader.java:362)
     [java] 	at java.security.AccessController.doPrivileged(Native Method)
     [java] 	at java.net.URLClassLoader.findClass(URLClassLoader.java:361)
     [java] 	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
     [java] 	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:331)
     [java] 	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
     [java] 	at org.usfirst.frc.team9999.robot.Robot.robotInit(Robot.java:42)
     [java] 	at edu.wpi.first.wpilibj.IterativeRobot.startCompetition(IterativeRobot.java:65)
     [java] 	at edu.wpi.first.wpilibj.RobotBase.main(RobotBase.java:198)
     [java] Caused by: java.lang.ClassNotFoundException: edu.wpi.first.wpilibj.command.TimedCommand
     [java] 	at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
     [java] 	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
     [java] 	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:331)
     [java] 	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
     [java] 	... 15 more
     [java] WARNING: Robots don't quit!
     [java] ---> The startCompetition() method (or methods called by it) should have handled the exception above.
     [java] WARNING|Gazebo Transport: Ignoring unsubscribe: topic: "/gazebo/frc/simulator/pwm/7"
     [java] host: "192.168.1.66"
     [java] port: 52976
     [java] msg_type: "gazebo.msgs.Float64"

Your assistence is greatly appreciated.

thank you

kbowen99, 2017-01-15:
I've Had the same issue. It is not an issue with the simulator, but an issue with the code you are running. If you write your own drive code it will most likely work. The error "Robots don't quit!" Means that your code exited (most likely crashed). You can read through the code and fix it, but rewriting the drive part is much easier.

andrasfe, 2017-01-15:
Thanks for your input. Just wanted to highlight the fat that, as I mentioned the original issue, I did not write any code. This is the initial experience I got when running the sample code provided, through Eclipse.

sciencewhiz, 2017-05-09:
I think this is because screensteps (http://wpilib.screenstepslive.com/s/4485/m/23353/l/228979-installing-frcsim-manually-ubuntu) points you to installing the 2016 simulation library (http://first.wpi.edu/FRC/roborio/maven/release/edu/wpi/first/wpilib/simulation/simulation/1.0.0/simulation-1.0.0.zip) instead of the 2017 library (http://first.wpi.edu/FRC/roborio/maven/release/edu/wpi/first/wpilib/simulation/simulation/2017.3.1/simulation-2017.3.1.zip) .

sciencewhiz, 2017-05-17:
I confirmed that when I used the 2017 simulation library, I didn't get those errors.

D0048, 2018-01-18:
Experiencing the same using the 2017 simulation lib, could someone help?

williamhaley, 2018-02-08:
@D0048 I've been struggling with this too and encountered a number of issues trying to get steps which worked for me a month ago to work properly right now. I was able to run everything against the simulation environment late 2017, but when I tried again in the past couple weeks I encountered various issues. I eventually got it working, but it was an involved and time consuming process.

Apologies if I'm way off base here (I'm making a lot of assumptions on _why_ certain changed worked for me, plus there were a lot of moving pieces so maybe some changes I made _seemed_ to help things but were coincidental), but this is what eventually worked for me...

1. Gazebo 9 was released a couple weeks ago. I found I had to downgrade to Gazebo 8. You can edit the [one-liner install script](http://get.gazebosim.org) to use `8` instead of `9`. No idea why I needed this. Seems the Gazebo includes dev files and perhaps the latest are incompatible? Either way, I saw build errors for the simulation gradle plugins job until I did this.
1. The latest Eclipse plugin (as of time of writing) is `v2018.2`, which seems to be incompatible with the `v2017.3.1` FRCSim files. The [latest simulation files I found](http://first.wpi.edu/FRC/roborio/maven/release/edu/wpi/first/wpilib/simulation/simulation/) are for `v2017.3.1`, nothing for 2018. I had to manually install an [older version of the Eclipse plugins](http://first.wpi.edu/FRC/roborio/release/) to get a matching version. Perhaps the Eclipse plugin isn't backwards compatible? Either way, I saw runtime errors until I downgraded.
1. I had to install a 2017 build of the `frc-toolchain` - https://launchpad.net/~wpilib/+archive/ubuntu/toolchain for `yakkety` on my 16.04 box
1. I had to use tag `v2017.3.1` from `allwpilib` to build the simulation plugins. I won't pretend to understand the nature of the changes in `allwpilib` between `v2017.3.1` and `v2018.1.1`, but they seem significant - https://github.com/wpilibsuite/allwpilib/compare/v2017.3.1...v2018.1.1 The README for this repo says you can build simulation files with `./gradlew build -PmakeSim`, but I only found that worked with `v2017.x`

With all that, things worked and I can run some sample Robot code against a sample Robot Gazebo model and control it 🎉 , but in my experience it's very daunting to export CAD models of _your_ robot design into the simulation environment. I'd love to hear if other teams have done it and any steps they can recommend on how it's best done, but the process seemed too extensive for our team to tackle this year.

Don't get me wrong, it's certainly helpful to have _any_ simulation environment where your code can execute, but I've encouraged our team to test against the physical robot this year due to our limitation when it comes to getting our robot into the environment.

D0048, 2018-02-08:
@williamhaley Thanks a lot! 

Btw, I just sadly learnt that the frc_sim project is not supported for 2018, which makes simulation for the 2018 version of WPILib API under OS like Linux impossible. The older version still works well thou... I guess our team will have to go with depreciated APIs for simulation, and migrate to the newer ones once the physical robot got fully constructed.

AustinShalit, 2020-03-10:
At this time we do not have plans to continue supporting the Gazebo simulator. There are members of the team working on low fidelity simulation for teams  to validate robot code. If someone in the community wants to take that project on, please open an issue so we can discuss.

closed 2020-03-10 by AustinShalit