wpilibsuite/allwpilib #473: GRIP Code Generation Capitalization Error

RaidingForPants, 2017-01-27:
When using the GRIP Computer Vision Pipeline to generate code, the function which overrides the virtual function "Process" in the parent class, VisionPipeline, is not capitalized, and is generated as "process".

In MyPipeline.h:
`public:
		MyPipeline();
		void process(cv::Mat source0);
		cv::Mat* gethsvThresholdOutput();
		cv::Mat* getcvDilateOutput();
`

In VisionPipeline.h:
`public:
  virtual ~VisionPipeline() = default;

  /**
   * Processes the image input and sets the result objects.
   * Implementations should make these objects accessible.
   */
  virtual void Process(cv::Mat& mat) = 0;
`

333fred, 2017-01-27:
This issue was moved to WPIRoboticsProjects/GRIP#805

333fred, 2017-01-27:
Moved to https://github.com/WPIRoboticsProjects/GRIP/issues/805.

closed 2017-01-27 by 333fred