wpilibsuite/allwpilib #478: I2C methods return "bad" even if "good"

tom131313, 2017-02-09:
I2C methods (involves files I2C.cpp and HAL\I2C.cpp) check for lower level functions returning the number of bytes processed and indicate error if the requested number doesn't match the processed number.  Various methods return error even if there is no error so apparently the assumption that the lower level functions return other than 0 or 1 is incorrect.

calcmogul, 2017-02-09:
i2clib provided in the NI libraries has no documentation whatsoever, so I had to guess as to what the return value meant. Based on the testing I did with an ITG3200 gyroscope, which returned 2 on successful reads of 16 bits and -1 on failure, it should work as the comment on HAL_TransactionI2C() describes: `@return The number of bytes read (>= 0) or -1 on transfer abort.`

Based on this assumption, wpilibc handles the return value from the HAL properly. If `status < receiveSize`, either not enough bytes were received or -1 was received. In either case, the wpilibc function should return failure. Note that it returns false for success and true for abort for compatibility with what the documentation previously claimed should happen.

tom131313, 2017-02-10:
Thanks for your quick look at this issue.

We are getting back a return code of 1 even for 2 bytes returned from the
ITG3200.

The code that fails this year because of the bad status worked fine for
the 2016 season.

I'd like to run your test code, if you can send it to me.

What version of the Lidar Lite are you using?  Ours is Blue Label
manufactured 8/2015.

Thanks,

Rick

Richard K. Thomas

tom131313, 2017-02-10:
Sorry - battling computers all day and fuzzy late at night.

Neither our Lidar Lite nor our ITG3200 work this year and both worked
last year.

I would like your ITG3200 test code to run.

We use the 6DOF combo board with the ADXL345 and ITG3200.

Thanks,

Rick
=========

On Thu, 9 Feb 2017 22:45:32 -0500 <bike13@juno.com> writes:
Thanks for your quick look at this issue.

We are getting back a return code of 1 even for 2 bytes returned from the
ITG3200.

The code that fails this year because of the bad status worked fine for
the 2016 season.

I'd like to run your test code, if you can send it to me.

What version of the Lidar Lite are you using?  Ours is Blue Label
manufactured 8/2015.

Thanks,

Rick

Richard K. Thomas

Richard K. Thomas

calcmogul, 2017-02-10:
That's odd, because the code from 2016 was even more unreliable than it is now before I changed it and didn't match the behavior described in the comments that WPILib was depending on.

I used the breakout board from https://www.sparkfun.com/products/11977 and a custom gyro class: https://github.com/Team3512/RobotModules/tree/master/ITG3200_Gyro. I added print statements to WPILib's HAL/I2C.cpp and watched the output over time while calling `GetAngleX()`.
[itg3200_gyro.tar.gz](https://github.com/wpilibsuite/allwpilib/files/765653/itg3200_gyro.tar.gz) contains some test code, although it's not the same code I was using.

tom131313, 2017-02-10:
hello

You think "this is Odd!?"  Nicely understated!

I was hoping you had definitive NI documentation that explained what was
happening with the NI functions.  My 2 issues submitted were intended to
give strong hints of where the problems may lay but weren't 100%
rigorously correct and thorough.  Lacking good documentation I have done
more testing to figure out better what is happening (not a good
programming technique but apparently all we are going to get).

I have an I2C ITG-3200 gyro on a 6DOF combo board with an ADXL345 (
https://www.sparkfun.com/products/10121 ).  I have a Lidar Lite Blue
label.  These devices can test many but not every combination of I2C
functions.

The HAL_TransactionI2C returns a status of 2 if reading 1 byte or 8
bytes.  So Read thinks reading 1 byte succeeded and reading 8 bytes
failed.  Actually both transactions succeeded from a device functionality
point of view - they worked okay for their intended purpose.  It looks
like status < 0 would be a good return code.

Note that AddressOnly uses Transaction with null pointers and zeros. 
That usage always returns a -1 from HAL_TransactionI2C if a device is
connected or not.  Thus AddressOnly is a useless function maybe because
of how Transaction works (badly?) with nulls and zeros.

The Write function call to HAL_WriteI2C correctly says to write 2 bytes
(sizeof(buffer)) and HAL_WriteI2C returns a status of 1.  Write then
computes return pass/fail incorrectly.  It thought the return from HAL
should have been sizeof(buffer) (a 2) which it wasn't.  Write returns a 1
thus appearing to fail.  Using a return status != 1; may be good.

WriteBulk returns a 1 if 1 byte is written.  That test is inconclusive
but all I could do with my devices.

HAL_ReadI2C called by ReadOnly returns 1 for 1 or 2 bytes read so that
pass/fail needs to be fixed also.

I'll try to run your ITG-3200 test soon.  I don't know what you meant by
itg3200_gyro.tar.gz (there was no attachment) but I'll try the program I
see in Github.

Thanks,

Rick
PS - In this note I've ignored the other issue I submitted - gyro won't
work without the fix - the problem Read has telling Transaction sendSize
sizeof(registerAddress) which is incorrectly a 4 instead of the required
1.  (Gyro dies a horrible death curable only by cycling the power if a 4
is used instead of a 1.)
PPS - what's really odd is the ADXL345 on my combo board has data
starting in register 0x30 and not 0x32 like the WPILib and data sheet
say.  I decided not to fuss about that problem as we use other
accelerometers now.
======================

Richard K. Thomas

tom131313, 2017-02-10:
hello

I have run the gyro code from github as you suggested:
https://github.com/Team3512/RobotModules/tree/master/ITG3200_Gyro

I understand what you are seeing.

An unfortunate co-incidence fools us into thinking Transaction status = 2
is the number of bytes because that code only reads 1 direction at a time
- 2 bytes of data.  And it doesn't test all of the I2C functions.

My program reads all 4 pieces of data - temperature, x, y, z - at the
same time for a total of 8 bytes.  Transaction still returns 2 and that
might have been intended to be the number of bytes but it isn't now for
some unknown reason and we don't have NI's documentation or source code.

I'd be happy to share with you my work-arounds in I2C.cpp but I am
nervous that I don't know much about NI I2C and that should be tracked
down with NI for most reliable programming of I2C.

Thanks,

Rick
==========

Richard K. Thomas

calcmogul, 2017-02-11:
The attachment is on GitHub. I think I edited it in later and you weren't sent an update. That's irrelevant now though.

I sent an email to one of our NI contacts today asking for documentation.

jhershbe, 2017-02-13:
The functions in i2clib directly return the value returned by the ioctl() OS function. That means if you get -1, it's an error and you should check errno. Otherwise it was successful and the parameter buffers that were passed in have been updated.

PR #484 (2017-02-17): Corrects assumptions about return values from i2c-lib
Fixes #478
commit 12f75986 (2017-02-17): Corrects assumptions about return values from i2c-lib (#484)

Fixes #478
closed 2017-02-17 by PeterJohnson