wpilibsuite/allwpilib #480: JVM on Robot Core Dumps when Robot Code Starts

adamzg, 2017-02-10:
After deploying new code to the robot we are experiencing occasional core dumps.  The JVM will keep restarting until the JVM bootstraps the robot program successfully.  

Attached are the core dumps we've experienced on 2 different roborios tonight.

[CoreDump 2-9-16.zip](https://github.com/wpilibsuite/allwpilib/files/765554/CoreDump.2-9-16.zip)

PeterJohnson, 2017-02-10:
We will need your robot code to try to help debug this issue.  The stack traces don't really contain enough info.

333fred, 2017-05-07:
@adamzg ping?

adamzg, 2017-05-08:
Sorry for the delay.  The team has opted not to continue pursuing the
origins of the bug further having made changes in the code that avoid the
issue.

czipperz, 2017-05-11:
What were the changes?

adamzg, 2017-05-11:
I don't have access to the code anymore.  Keith may be able to help you.

adamzg, 2017-05-14:
Our problem went away after some changes. I could not relate any of the changes to what made core dumps go away.

 

Keith

 

From: Adam Gresh [mailto:adamgresh@gmail.com] 
Sent: Thursday, May 11, 2017 7:32 AM
To: wpilibsuite/allwpilib; Keith Glasnapp
Subject: Re: [wpilibsuite/allwpilib] JVM on Robot Core Dumps when Robot Code Starts (#480)

 

I don't have access to the code anymore.  Keith may be able to help you.

sciencewhiz, 2017-09-04:
Closing as there isn't enough info to reproduce or debug

closed 2017-09-04 by sciencewhiz