wpilibsuite/allwpilib #485: Incorrect low resolution for Joystick axes

tom131313, 2017-02-17:
We have a TI LaunchPad (as a USB connected joystick to the Driver Station PC) with a potentiometer that returns 0 to 4095 through its full range of motion for each axis.  This can be verified on the Microsoft Windows calibrate in Setup USB Game Controllers.

Using classes Joystick or DriverStation the robot code receives the axes positions as -1 to +1 as expected but the steps are very course - about 256 steps for the range instead of 4095 steps.

It is hard for us to troubleshoot any better than having a strong conjecture that the problem is in HAL file **FRCDriverstation.cpp**.

The comment ```// copy integer values to double values``` is reasonable but it appears the code actually converts ```int16_t``` to ```int8_t```, divides by ```127.``` with conversion to ```float```.

This problem greatly diminishes the utility of the potentiometer for fine tuning motor performance, for example.
```
int32_t HAL_GetJoystickAxes(int32_t joystickNum, HAL_JoystickAxes* axes) {
  HAL_JoystickAxesInt axesInt;

  int retVal = FRC_NetworkCommunication_getJoystickAxes(
      joystickNum, reinterpret_cast<JoystickAxes_t*>(&axesInt),
      HAL_kMaxJoystickAxes);

  // copy integer values to double values
  axes->count = axesInt.count;
  // current scaling is -128 to 127, can easily be patched in the future by
  // changing this function.
  for (int32_t i = 0; i < axesInt.count; i++) {
    int8_t value = axesInt.axes[i];
    if (value < 0) {
      axes->axes[i] = value / 128.0;
    } else {
      axes->axes[i] = value / 127.0;
    }
  }

  return retVal;
}
```

ThadHouse, 2017-02-17:
This is actually a limitation of the DriverStation. The DS sends all joysticks as 8 bit values over the network, so that is what we decode them as. The int8_t conversion isn't actually what is causing the issue.

ThadHouse, 2018-05-14:
Do we think the DS API is going to change regarding this. I doubt it, and would like to close this until that happens.

tom131313, 2018-05-15:
It's disappointing not to take full advantage of devices such as the TI
LaunchPad that FIRST has encouraged us to try.

Closing this issue is reasonable if that root cause dependency on the DS
will never be addressed.

Richard K. Thomas

AustinShalit, 2020-03-10:
Closing due to bug in dependency.

closed 2020-03-10 by AustinShalit