wpilibsuite/allwpilib #486: PIDController: ClearIAccum and SetIZone

Bubba74, 2017-02-17:
After looking through the PIDController header file, I realized that there is no function to clear the accumulated error, nor to set a zone outside of which the Integral value has no effect. These functions seem simple to implement and largely beneficial in all sorts of scenarios. But without these functions, I'm looking at either skipping an Integral value (which would severely hinder the controller) or implement some awkward work around.

I've read some stuff on additions to WPILib and it says that no additions will be made during the build season. This makes perfect sense, but I am wondering, before that might happen, how best to implement this functionality. Right now, I have modified the PIDController header file to make certain variables public as well as the Calculate function, so that I might modify these readily. I suppose the surest way of implementing this would be to modify the PIDController.cpp file and then rebuild WPILib, but I would prefer not having to rebuild the entire library.

I'd appreciate any pointers as to what I might do to implement these two functions without messing around too much with the WPILib source.

Thanks in advance,
Henry Loh

bradamiller, 2017-02-17:
If you look at the allWPILib repository there is a pull request to do what you want and the java version has been tested on a real robot https://github.com/wpilibsuite/allwpilib/pull/482. If you grab the code and apply the pull request, then you can take the PIDController class and add it to your program. We will do a release with that incorporated, but it didn't make it into todays release.

Bubba74, 2017-02-17:
Thanks so much. So is the I-Zone planned to be implemented in next year's release?

closed 2017-02-17 by Bubba74

bradamiller, 2017-02-18:
I expect it will go out in the next release that we do. It was put it in a few days ago and we really haven’t had time to test it thoroughly although at least one team is running with the Java version right now.

Bubba74, 2017-02-18:
I imagine there shouldn't be any problems with the addition, I looked at the Pull Request and the changes were so minimal, but I suppose one can't just update wpilib without thorough testing.

Thanks for the notice.