wpilibsuite/allwpilib #487: WPILibVersion reporting incorrect version

jhh, 2017-02-17:
It appears the version string declared in the 2017.3.1 `WPILibVersion.h` is defined as "2017.1.1". I double checked the installed version and the header file comment says the version string is auto-generated:

```c
/*
 * The corresponding WPILibVersion.cpp file is autogenerated by the build
 * system. If it does not exist, make sure that you run a build.
 */
extern const char* WPILibVersion;
```

PeterJohnson, 2017-02-18:
I just tested this with the latest eclipse plugins and it's printing the correct version (and the DS is showing the correct version).  Are you using the Eclipse plugins to deploy your code to the robot?  The version is stored in the wpilibc.so shared library.  The Eclipse plugins copy this and other libraries to the robot; if you're using some other method to deploy your code, make sure it is copying the libraries as well.

jhh, 2017-02-18:
Non-eclipse was my problem, thx for the help.

closed 2017-02-18 by jhh