wpilibsuite/allwpilib #490: Broken NavX Micro Over I2C MXP

bigabos-hey, 2017-02-20:
After upgrading to 2017.3.1, I2C communication with NAVX micro failed.  USB communication still functioning.  Roll back to 2017.2.1 resolved issue.

kauailabs, 2017-02-23:
I've looked into this, and w/2017.3.1, the I2C ReadOnly function (verified in Java but also believed to bein C++ as well) returns and no data is filled into the read buffer. There's a write that occurs immediately before it.

I"m going to break out the logic analyzer next, but something appears to have changed in 2017.3.1 that is causing I2C communication that's worked for a long time to break.

Please let me know if any more information is needed - I'll post info from the Logic Analyzer when I have some.

PeterJohnson, 2017-02-25:
There were only two changes to I2C between 2017.2.1 and 2017.3.1:

https://github.com/wpilibsuite/allwpilib/commit/5a5f10dfc8a57295d86092efea9faabe2e392a49
(which affected only the C++ I2C::Read() function)

https://github.com/wpilibsuite/allwpilib/commit/12f759860eea3127d3ff1ccabbf801a31ffef863
(which affected all return values, but should not have affected any actual operation)

kauailabs, 2017-03-03:
Thanks, Peter.  Regarding:

12f7598 (which affected all return values, but should not have affected any actual operation)

While I realize this change brings the implementation in alignment w/the documentation, this change actually did impact act actual operation, in that the sense of the return values which have been in WPI Library for the last 2 years has changed; we've recently updated the navX-MXP library to match.

I mention this because there may be teams using the I2C APIs to communicate w/other devices that could be impacted as well.

@bigabos-hey, I'd like to recommend this item now be closed, after you've had a chance to test the updated (v. 3.0.329) navX-MXP/navX-Micro RoboRIO Library.

bigabos-hey, 2017-03-04:
We'll check it out later tonight.

Thanks

closed 2017-03-08 by bigabos-hey