wpilibsuite/allwpilib #491: SendableChooser (ITable) data is unordered

rtrahms, 2017-02-20:
Because a hashtable is used to store the key/object pairs, the table order is basically random.  This issue is to request changing this to an ordered data structure.

ThadHouse, 2017-02-20:
What issues are you running into with this being unordered? And if we were to attempt to order it, what criteria would we use to sort? In addition, random access of a hashtable is likely quicker than a search and select from an ordered list.

rtrahms, 2017-02-20:
For a large collection of autonomous modes, it becomes difficult to find the right one when they are all dumped randomly on the smart dashboard.  We are working around this by creating multiple choosers each with a smaller set of items, but it would be nicer not to have to do this.

Regarding the quicker search, I agree on the general benefit hashtables bring, but am not sure why you would need a quick search on a small group of items.  It seems like the wrong type of application for this data structure, given the nature of smaller collections of objects.

rtrahms, 2017-02-20:
Regarding criteria for ordering, I think it is fairly simple - report them out in the order they were entered.   A list, arraylist or similar ordered data structure is fine for this.

trc492, 2017-03-11:
I second that request. We have implemented a mechanism similar to SendableChooser in FTC called FtcChoiceMenu and it is presenting the choices in the order they are added.
https://github.com/trc492/FtcSamples/blob/master/Ftc3543Lib/src/main/java/ftclib/FtcChoiceMenu.java

cyocom, 2017-03-22:
If you switch the Java from the HashMap Map implementation to LinkedHashMap implementation it will preserve the keyset order. [See the javadoc here.](http://docs.oracle.com/javase/8/docs/api/java/util/LinkedHashMap.html)

trc492, 2017-03-22:
BTW, there is a forum thread discussing this issue and I have added my comment to it.
https://www.chiefdelphi.com/forums/showthread.php?t=155638
My comment is repeated here:
I don't know why they use hash table either because the number of objects in a SendableChooser are generally small. Hash table is good for performance dealing with large number of objects. Whatever reason that was, they can still use hash table. They just need to somehow store the object order as part of the info in the hash table. For example, they could prefixed the name string with the order number before storing it into the hashmap. When displaying on the SmartDashboard, they could sort the collection by "name" since the name prefix is the order number. Then before displaying on the dashboard, they could strip out the prefix.
But of course, if there is no good reason to use hashmap, then they should go back to using some sort of ordered list.

PR #492 (2017-05-07): SendableChooser Structure HashMap -> LinkedHashMap
closes #491 

~SmartDashboard will also need an update.~ SmartDashboard respects the array order already.

- [x] Java
- [ ] C++
closed 2017-05-07 by 333fred