wpilibsuite/allwpilib #497: DriverStation::GetMatchTime() emulation returning wrong value type

MrRoboman4321, 2017-03-21:
When calling GetMatchTime() without the robot being connected to the FMS, the documentation specifies that DriverStation attempts to emulate the behavior as if it was through the use of practice and autonomous mode enabling. When you run it without being connected, a float value is returned. However, when in a match connected to the FMS, it will return an integer value.

333fred, 2017-05-07:
@PeterJohnson are we sure this is a bug in the dependency, or a bug on our end?

333fred, 2017-05-07:
Actually, scratch that. I think it might be time for me to go to bed.

ThadHouse, 2018-05-14:
The FMS internally definitely uses integers for match time (And I believe the FMS protocol is integer as well). We have a few options for this. The FMS could change to floating point, the DS could always round to integer before sending, or we could mask to integers in the HAL. But I would like to do one of the 3 so we can close this issue. Most likely making the DS truncate to integers is likely the best option.

Starlight220, 2023-01-15:
Five years, has anything changed for this?

closed 2026-11-14 by PeterJohnson