wpilibsuite/allwpilib #500: Request: add support for using maximum error in PIDController buffer when computing onTarget() instead of average error

gartaud, 2017-03-30:
Background:
We are expecting onTarget() to return true when the system is on target and has settled there.
We therefore feel that the current implementation of onTarget() is problematic because if used with a tolerance buffer of default size 1 the method will report true if we are at the target at a random instant even if the system is still settling.
Using a larger tolerance buffer should address the issue, but the current implementation uses the average of the buffer content rather than the maximum. This is not very helpful as the average could be on target even if the system is still oscillating.
Request:
We therefore respectfully request that you consider adding an option to use the maximum error rather than the average as an option.
With that option enabled onTarget() would return true only if all the measured errors in the buffer are within the tolerance.
Thank you.

calcmogul, 2017-03-30:
The PIDController class uses the buffer for a FIR filter to filter noisy input signals (e.g., velocity estimates), so the current averaging behavior reflects that. Since intermediate values of the filter are noisy, only the output of the filter should be trusted. PIDController should be using a LinearDigitalFilter instance for this instead of a raw buffer, but the patch for that isn't passing the integration tests yet.

The real problem is what PIDController's onTarget() function considers to be steady state (i.e., the output staying at the reference versus oscillating around it). Since PID controllers have a second order response, it would be sufficient to check that the change in filtered error is close to zero in addition to the filtered error. A defaulted argument could be added to SetTolerance() for specifying a tolerance for it.

gartaud, 2017-03-30:
Hi @calcmogul Thank you for your comment. You are right, the real problem is what PIDController's onTarget() function considers to be steady state. Checking that the change in filtered error is close to zero would also work for us. I find it hard to believe that is not much more complaining about the current implementation.

PeterJohnson, 2019-07-30:
Fixed by #1300.

closed 2019-07-30 by PeterJohnson