wpilibsuite/allwpilib #501: Better bebugging capabilities

zero-frost, 2017-04-01:
This isn't actually an issue, more of a request. At the moment, roboRIO debugging is limited to GDB. This would be fine, however what would be nice is having a way of calling into HAL to read values out (e.g. PWM values, encoder counts). The reason I am bringing this up, is I was planning to write a GDB extension to read/write values directly, and wanted to know if it was possible. Writing would be tricky, as you need to reroute a function call en route.

333fred, 2017-05-07:
The roboRIO isn't really limited to GDB, it's limited to whatever has been compiled for gcc4.9-armV7-sflt. I don't think any of us are particularly familiar with writing GDB extensions, so while we might be able to give you some help, I'm not really sure what you mean by rerouting function calls.

Kevin-OConnor, 2018-06-08:
"a way of calling into HAL to read values out". Isn't this what the existing HAL APIs do? How is what's being requested different from the HAL APIs that already exist?

AustinShalit, 2019-08-04:
Closing because this can be done with the existing HAL APIs.

closed 2019-08-04 by AustinShalit