wpilibsuite/allwpilib #511: SendableChooser elements are randomly ordered when displayed on SmartDashboard

czipperz, 2017-04-26:
This affects the Java version.  Not sure if it affects the C++ version.

AustinShalit, 2017-04-26:
Duplicate of #491

czipperz, 2017-04-26:
@AustinSchuh 
Looking at the C++ source, I'm not sure where it adds values to the table member. https://github.com/wpilibsuite/allwpilib/blob/master/wpilibc/shared/include/SmartDashboard/SendableChooser.inc#L29
In the Java version it is explicitly there. https://github.com/wpilibsuite/allwpilib/blob/master/wpilibj/src/shared/java/edu/wpi/first/wpilibj/smartdashboard/SendableChooser.java#L65

czipperz, 2017-04-26:
I think for the Java part it is easily fixable just by storing the `m_map.keySet().toArray(new String[0])` as a variable and sorting it using a `Collator`.

I don't know where to look in the C++ source.

333fred, 2017-04-26:
Closing as duplicate.

closed 2017-04-26 by 333fred