wpilibsuite/allwpilib #514: Why do the builtin commands use _ prefixes for their actions?

czipperz, 2017-04-28:
CommandGroup implements its logic in `_initialize`, `_execute` and so on.  The reason this is bad is because it prohibits teams from using the decorator pattern with builtin commands.  This will not work:

    public class CG extends CommandGroup {
        public void initialize() {
            // do custom initialization here that must occur before CommandGroup is initialized.
            {
                addSequential(...);
                //...
            }
            super.initialize();
        }
    }

Our team tried to do this in order to allow for member variables to be initialized at runtime that were `private`.  We ended up being forced to make them `public` and expose our implementation.

calcmogul, 2017-04-28:
I'm told the original reason for this was so inexperienced teams couldn't forget to call super.initialize() and break things by accident. Personally, I prefer users doing things correctly instead of the library bending over backwards to accommodate ignorance with regard to the language and tools. Granted, this is coming from a veteran C++ guy. The teams I interact with have the luxury of a software mentor to catch potential mistakes like that, while the average FRC team likely doesn't.

czipperz, 2017-04-28:
Ok cool.  I'm going to give yall a PR for this.  Its a super simple change it turns out.

calcmogul, 2017-04-28:
Wouldn't requiring calling super.initialize() at this point break any classes written by teams that subclass CommandGroup?

czipperz, 2017-04-28:
This is a silent breaking change as you mention.  Could it just throw an exception if `initialize()` isn't called when `execute()` happens?

czipperz, 2017-04-28:
And then we could release a major patch that removes the unnecessary memory half way through build season?

calcmogul, 2017-04-28:
As a policy, we only make bug fixes during build season.

calcmogul, 2017-04-28:
If we do break user code, we need to decide whether the change is worth it. We made decisions like that in 2016 when modernizing the code base. This one seems rather minor for a lot of breakage. If you can figure out a way to introduce the change gradually over the course of two years (that is, in the same way deprecation and removal occurs), I'd feel much better about it.

czipperz, 2017-04-28:
Well to be clear we don't have to ever remove the unnecessary memory needed to store whether `initialize()` was called before `execute()`.  Then the breaking would be explicit when teams run/test the code because it throws an error.

calcmogul, 2017-04-28:
Could you point me to some code exhibiting the issue this solves? I may have missed something and don't understand your problem fully. For example, what variables are you having to make public?

The issue is not silent vs loud breakage, although the latter is preferrable. It's that breakage is being caused for a lot of teams for what I'd argue is not that beneficial for them.

czipperz, 2017-04-28:
"a lot of teams".  I don't think either of us know how many teams override these methods without calling `super.`.  I understand this may sound rediculous, but can you find a single example of any team that does this?  I've looked through the top ten results of teams of GitHub by `frc` and the only one that overrode these functions called the super version: https://github.com/Team694/Rafael/blob/bf3432e7759d9e035e72a2cb5d007b68f3d96d6a/src/com/stuypulse/frc2017/robot/commands/cv/SetupForGearCommand.java

calcmogul, 2017-04-28:
This was the first result for me for `FRC command java`: https://github.com/mr-glt/FRC-2017-Command/blob/master/src/main/java/org/usfirst/frc/team6027/robot/commands/WinchToggle.java. It inherits from Command and doesn't call any of the superclass methods. Hardly a representative sample though.

czipperz, 2017-04-28:
Just to be clear: inheriting from Command and not calling the super methods and inheriting from CommandGroup and not calling any super methods are two different things.  The one in which you do not call the super commands of Command is fine because they used to be `abstract` / `virtual = 0` and are now defaulted to do nothing.

However inheriting from CommandGroup is what I was asking for.  I am currently on page 3 of frc results and intend to go at least until page 20.

calcmogul, 2017-04-28:
Good point. The code I've seen so far only uses CommandGroup to order autonomous commands.

With that said, I don't like the idea of keeping a warning in the code base for all time. On the other hand, if we removed it at some point, old code teams copy-paste from previous years could break silently from then on. hm...

czipperz, 2017-04-28:
Here is a team that does it https://github.com/Spartronics4915/2016-Stronghold/blob/11866539e1a5a6f8e276c52a9e9a4621ac3cf410/src/org/usfirst/frc/team4915/stronghold/commands/AutoCommand1GP.java

czipperz, 2017-04-28:
I did not mean making _initialize public although that might fix this issue. Your interpretation was correct before the edit. I meant making one of my member variables public.

What we ended up doing was making an interface called Initable that declared init(). If the chosen command from the SendableChooser was Initable, we call init() before calling​ start(). I only knew of the _ methods because of looking at Command.java's source. I would've liked for our team to be able to just do initialization in initialize() like intended. Removing the _ methods would do this

calcmogul, 2017-04-28:
If we make them public, we should give them better names than `_initialize` before doing so. It's not intuitive what the functions are for from their names, and I'm not sure what a good naming scheme would be off the top of my head.

Yeah... we painted ourselves into a corner in that regard. As much as I'd love to fix it, I don't think doing so would be OK. I'd have to consult some of the other devs about it.

333fred, 2017-04-28:
I'm not of the opinion that the logic should be overridable. As has been pointed out, this is too easy for teams to accidentally break. We should be trying to steer teams towards success, rather than failure here. We should be using the pattern of having something like
```java
void _initialize() {
    initialize();
    // Existing initialize code
}
```
---
Now, that being said, we also have to consider how we can not confuse teams with naming here. I see a few options:
1) Move the existing initialize to be as described above. This is a breaking change, but shouldn't be particularly bad. We'd have to make a few small changes to a couple of our commands that override `_initialize`, but this wouldn't be too bad of a change. Most importantly, it shouldn't break most team's commands, since they don't have to call super.
2) Create a new method. This wouldn't be good, I think. Intuitively, what comes first? `initialize` or whatever else we make? Not a fan.
---
Another point: @czipperz, the example you pointed out isn't going to work with the way that most people start commands, namely by calling `start` on the command (for example, this is how buttons start commands). By the time either `initialize` or `_initializeCommand` has been called, the `start` codepath has set the `m_locked` variable, and any calls to `addSequential` or `addParallel` are going to throw an exception. I think the problem you're trying to solve here is this: I'd like to add things to the command group at initialization time, which was, I thought, the point of the newly-added `ConditionalCommand`. Please correct me if I'm assuming anything wrong about this.

AustinShalit, 2018-05-30:
I think this question has been answered.  If you have further questions about this question, please reopen.

closed 2018-05-30 by AustinShalit