wpilibsuite/allwpilib #516: Why is the general style to have variable declarations outside of for loops where they belong?

czipperz, 2017-04-28:
Example:

    int16_t i;
    for (i = 0; i < size; i++) {
        ...
    }
    // I am sure `i` is not used in this line so it should be placed in the for loop
    return HAL_kInvalidHandle;

I am sure `i` is not used past the for loop body, a valid reason to have the declaration outside of the for loop.

The style guide does what I suggest.
Outside the loop when it should be for cleanup purposes https://github.com/wpilibsuite/styleguide/blob/master/cppguide.html#L978
Inside the loop https://github.com/wpilibsuite/styleguide/blob/master/cppguide.html#L1160 https://github.com/wpilibsuite/styleguide/blob/master/cppguide.html#L1899

calcmogul, 2017-04-28:
That's an oops. Feel free to fix it. :)

czipperz, 2017-04-28:
Places where it uses the weird style:

https://github.com/wpilibsuite/allwpilib/blob/master/hal/include/HAL/handles/LimitedClassedHandleResource.h#L62

https://github.com/wpilibsuite/allwpilib/blob/master/hal/include/HAL/handles/LimitedHandleResource.h#L58

https://github.com/wpilibsuite/allwpilib/blob/master/wpilibc/shared/src/Commands/Subsystem.cpp#L53

czipperz, 2017-04-28:
There actually aren't that many

ThadHouse, 2017-04-28:
The handle ones had reasons in the past. Those went through like 5 iterations, so not surprising we missed that. But yeah PR that if you can :)

czipperz, 2017-04-28:
@ThadHouse C89 is rough man.

333fred, 2017-05-07:
Closed in https://github.com/wpilibsuite/allwpilib/pull/517. Thanks for the contribution @czipperz!

closed 2017-05-07 by 333fred