wpilibsuite/allwpilib #518: Disabled Periodic never called on field

Kevin-OConnor, 2017-04-29:
DS does an IPConfig /release when set to DHCP when it detects end of match. This means few or potentially no packets go out with disabled mode meaning disabled periodic never gets called. The DS is probably the spot to fix this but capturing so it's not forgotten

AustinShalit, 2018-05-30:
Do we know if NI fixed this in the DS?

Kevin-OConnor, 2018-05-30:
It was definitely not changed in the DS. I don't think I ever mentioned it to Greg.

auscompgeek, 2019-05-19:
Given IterativeRobot is deprecated in favour of TimedRobot, is this actually an issue now?

Kevin-OConnor, 2019-05-20:
Oops, I should have closed this. I couldn't repro, DS waits for 5 disabled packets to detect match end which is 2 seconds. Should be plenty of time to send disabled packets to the robot.

closed 2019-05-20 by Kevin-OConnor