wpilibsuite/allwpilib #519: Add functionality that exists in LV to WPILibC/J

ThadHouse, 2017-04-29:
There is some functionality that exists in LabVIEW but not in WPILib. This issue will be a list of things that I find that need to be added.

- [ ] Set RIO power rail disable
- [ ] Limit Switch

ThadHouse, 2017-05-12:
I'll do Limit Switch and the RIO Power disable, but I think we need to discuss more how we want to implement disconnected periodic, especially with the new PeriodicRobot.

sciencewhiz, 2017-05-17:
This is what we did for disconnected: https://github.com/Beachbot330/BeachbotLib/blob/master/java/src/edu/wpi/first/wpilibj/BBIterativeRobot.java

Note that our use case was we wanted to run code with iterative before connecting. For our use case, we'll switch to Periodic/TimedRobot.

rzblue, 2023-08-01:
Power rail disable implemented in #5477

PeterJohnson, 2025-02-07:
OBE

closed 2025-02-07 by PeterJohnson