wpilibsuite/allwpilib #522: Double initialization detection

AustinShalit, 2017-05-06:
Improve reporting of initial instantiation (like a compiler).

czipperz, 2017-05-07:
What do you mean Austin?

sciencewhiz, 2017-05-09:
Currently, if a hardware port is instantiated twice, the robot crashes and reports the second place that it was instantiated. However, this means that the programmer has to go find the first instantiation. If the code remembered each instantiation, it could tell you both places it was used.

Here's an example: https://www.chiefdelphi.com/forums/showthread.php?t=157127

czipperz, 2017-05-10:
Out of curiosity, why is `Subsystem.h` in `/wpilibc/shared/include/Commands` instead of `/wpilibc/shared/include`?

sciencewhiz, 2017-09-04:
Anyone have any good ideas on how to implement this?

PeterJohnson, 2017-09-04:
It's not really possible in C++.  In Java we can grab the stack when we allocate a resource, look up the stack to find where the user code instantiated it, and store that associated to the allocated resource.  Then when the next allocation request for the same resource comes in we have the info to point to the previous location.

sciencewhiz, 2017-09-04:
That's along the lines I was thinking of. The problem is that the HAL is what is allocating the resource, and it doesn't make sense to pass information to the HAL that only helps one language. The problem with doing it the java code is how to handle when hardware was allocated in a different class. For example an MPX DIO that is allocated as a Encoder (DIO input), ultrasonic (DIO output) and a speed controller (PWM).

PeterJohnson, 2017-09-05:
Well, it might help other languages, e.g. Python, just not C++.  I would think what would would pass to the HAL would just be a location string.  Alternatively we could do it at the JNI layer rather than HAL proper.

PR #3322 (2021-05-01): [hal] Report previous allocation location for duplicate indexed resource
Closes #522

This will allow us to pass a stacktrace down to know where previous allocations occured
closed 2021-05-01 by ThadHouse