wpilibsuite/allwpilib #536: hal::priority_mutex backwards compatibility shim breaks when using namespace hal

ThadHouse, 2017-05-12:
I was noticing during testing that if I ever had `using namespace hal` in a file, the compiler would complain about not being able to differentiate between priority_mutex and hal::priority_mutex. Either removing the shim, or removing the using namespace fixed the issue. Anyone with a little more knowledge about C++ know how to fix this?

```
#include "HAL/cpp/priority_mutex.h"
using namespace hal;

priority_mutex myMutex;
``` 

throws error cannot differentiate between priority_mutex and hal::priority_mutex.

PeterJohnson, 2017-05-12:
Is the shim a typedef or a "using"?

ThadHouse, 2017-05-12:
It's `using priority_mutex = hal::priority_mutex;`

PeterJohnson, 2017-05-12:
Ah, that's an alias.  You want simply "using Hal::priority_mutex;"

ThadHouse, 2017-05-12:
Oh ok. I'll PR that soon.

PR #538 (2017-05-13): Fixes hal shim from alias to using item
Fixes #536
closed 2017-05-13 by PeterJohnson