wpilibsuite/allwpilib #540: Add unexposed FPGA functionality

ThadHouse, 2017-05-19:
While working on #519, I found some other FPGA functions that are not exposed to user program. If we can find somewhere to expose them, we should.

sciencewhiz, 2017-05-19:
What functionality?

ThadHouse, 2017-05-19:
I've found a few in Power so far, like more informative status information for the user rails, and ways to get voltage specific to things like the AIO outputs, the MXP outputs, and roborio power chip temperatures.

Starlight220, 2020-04-20:
In Java most of these should be exposed in the `RobotController` utility class.

rzblue, 2023-08-01:
Chip temperature is implemented in #5477

Gold856, 2023-08-04:
Now that #5477 is merged, is there anything else to add? Everything related to the power rails has been added.

PeterJohnson, 2025-02-07:
Closing due to 2027 controller change.

closed 2025-02-07 by PeterJohnson