wpilibsuite/allwpilib #590: Continuous Encoder Output with PIDController

karagenit, 2017-07-31:
I've run into an issue using a `PIDController` in Continuous mode with an `Encoder` as it's source - namely, the `PIDController` requires an input between the range set with `setInputRange()`, however `Encoder` has no way of limiting it's output to a certain bounds.

Let's take this example:

`setContinuous()` is true, our input range is set to [0, 100], our setpoint is 60, and our `Encoder` is reading 375.

In the `calculate()` method we call:

```
m_error = getContinuousError(m_setpoint - input);
```

which is passed to `getContinuousError()`:

```
protected double getContinuousError(double error) {
  if (m_continuous) {
    if (Math.abs(error) > (m_maximumInput - m_minimumInput) / 2) {
	  if (error > 0) {
	    return error - (m_maximumInput - m_minimumInput);
  	  } else {
	    return error + (m_maximumInput - m_minimumInput);
	  }
    }
  }

  return error;
}
```

We would pass `-315`, which would make `abs(-315) > 50` true and `-315 > 0` false, so we would return `-215` as our error. With proper wrap-around, we should see an error of 15 (375 -> 75). 

Three solutions suggested by @calcmogul :

###  Checking for input excursions in Encoder::Get() and constraining the output accordingly

In other words, adding the fields `m_continuous`, `m_minval`, and `m_maxval` to the `Encoder` class, having methods to set them, and checking to ensure that `get()` will always return a value between the bounds if the `Encoder` is in Continuous mode.

> The first option introduces branching control flow into Encoder::Get() where there previously was none, which will make people pay for something they aren't always going to use.

Granted, this is true - I'm not sure how big of a performance impact/concern this would actually end up being, as that's beyond my experience level.

### Using modulus on the feedback controller's error

This would work, sort of. I think if we always returned `Math.abs(error % max)` it would be fine, but using straight modulo/floor modulo wouldn't give us the right values all the time (-215 % 100 = -15) and (Math.floorMod(-215) = 85). 

This one is interesting, as it begets a modification not to `Encoder` but to `PIDController` itself - which could be good or bad. I'm not sure if having a continuous (aka wrap-around) feature for the `Encoder` will be useful anywhere else outside of a continuous `PIDController` - if it is, then we would want to go with a fix that modifies `Encoder` itself, not this one. However, if this *is* the only time this feature would really be used, then I think it makes more sense to add the feature to `PIDController` - not in the least because all it would require is calling `setContinuous()` and `setInputRange()` on only one object instead of two. 

Also, I'm not sure if it would be better to simply modify `getContinuousError()` (as suggested) or to add a new method such as `getContinuousInput()` which assures that the input value will be between the input range - with the second we could have a check such as:

```
if(m_continuous) {
  return (input % max + max) % max;
} else {
  return Math.min(Math.max(input, min), max);
}
```

which has the added benefit of limiting the input to the max/min regardless of Continuous mode.

### Using a Hall Effect sensor to trigger an interrupt that resets the Encoder

As I understand this, it isn't any different than using some Digital input as the Encoder's Index to routinely reset the encoder. However, I'm not sure that this would work: going back to our example with a max/min set to [0, 100], what would happen when the Encoder/shaft is spinning in reverse? Yes, the Encoder would be reset to zero when it reaches the sensor, but as it is moving in reverse it would immediately go below zero, and would be reset to zero again once it reached -100. This effectively makes the Encoder output on the range [-100, 100] which is outside of our desired range [0, 100], which leads me to believe that this wouldn't be enough and that additional logic would be required (unless, of course, I'm completely misunderstanding this proposed solution). 

*p.s. regarding actually implementing a minimum output as well - I'm pretty sure that just consists of using `max - min` in place of `max` for the different modulo calculations, and simply adding `min` to whatever output.*

sciencewhiz, 2017-09-04:
Based on the description in screensteps, and encoder that doesn't wrap isn't a continuous sensor. In that respect, doing the first option is the correct thing, as it makes the encoder act like the continuous sensor that PIDController expects.

http://wpilib.screenstepslive.com/s/4485/m/13809/l/599721-operating-the-robot-with-feedback-from-sensors-pid-control

```The PIDController object can also handle continuous rotation potentiometers as input devices. When the pot turns through the end of the range the values go from 5V to 0V instantly. The PID controller method SetContinuous() will set the PID controller to a mode where it will computer the shortest distance to the desired value which might be through the 5V to 0V transition. This is very useful for drive trains that use have continuously rotating swerve wheels where moving from 359 degrees to 10 degrees should only be a 11 degree motion, not 349 degrees in the opposite direction.```

karagenit, 2017-11-24:
Strange, never saw that definition of a "continuous sensor" on screensteps. Anyway, the changes I made in #601 shouldn't affect "traditional" continuous sensors like wrapping potentiometers, though it does modify the GetError method instead of the Get method... I thought this would be easier to implement, as there is already continuous-mode specific logic in the GetError method (whereas there isn't in the Get methods).

closed 2017-11-26 by karagenit