wpilibsuite/allwpilib #606: Implement Tri-state DIO.

JacisNonsense, 2017-08-16:
The digital IO pins on the RoboRIO are capable of acting in a tristate manner, combined with the internal 40k pullup resistor. This allows DIO to dynamically switch between Input and Output at any time. WPILib doesn't currently support this without some large overhead. 

Taking a brief look, it will require some work on the HAL side to make this happen, since direction is given at init time.

Adding some entry points to the HAL (perhaps with a lower latency, also) would make bitbanging certain protocols, as well as dealing with custom circuits, much simpler and efficient.

JacisNonsense, 2017-08-16:
This may relate to #540

ThadHouse, 2017-08-17:
Will work on this when the new build system is merged. The HAL function and the WPILib functionality should be easy. 

For the lower latancy API, what I'll do is provide an API that will provide a shared_ptr to the NI tDIO structure. This will allow direct access to the ni functions. These will be unsafe functions, and will not be documented on screensteps, as it becomes really easy to break all sensors on the robot just by doing something wrong.

ThadHouse, 2017-10-08:
So starting to work on this, and providing direct access to the tDIO structure and other structures is just not going to be possible. Access to those is controlled by a few mutex's, and attempting to update a value will easily cause race conditions. So I think we are not going to provide direct access. However, I do potentially think a HAL level bit bang api might potentially be a better solution, so if someone wants to work on that I'm happy to look at it. I was able to get the High Level API working however.

JacisNonsense, 2017-10-09:
@ThadHouse sounds good, let me know if your higher level api hits master and I'll work on some it some more (if needed) when the beta software is sent out

Starlight220, 2020-04-27:
The HAL part has been done in #835, so what's next? I can work on Java, but I think that doesn't help right now.

PeterJohnson, 2025-02-07:
Unclear if this will be a feature in the 2027 controller.  Closing.

closed 2025-02-07 by PeterJohnson