wpilibsuite/allwpilib #637: Clean up WPILibJ test execution and XML test result collection with Junit 5

JLLeitschuh, 2017-09-12:
Let me start by saying this probably shouldn't happen this year as the current system works so there's no big reason to change it.

# Contex

Back when I was writing the WPILibJ testing framework we were using Junit 4. Junit 4 has no mechanism built into it for it to produce XML output. TestNG which had support for running Junit 4 tests and producing XML was originally used as our workaround. However, we discovered that using`TestNG` to run tests didn't work for tests with some rules (eg. Repeat and Parameterized if I remember correctly).

As such, the way that I hacked around this was to remove `TestNG` include the entirety of [ANT and ANT-Junit](https://github.com/wpilibsuite/allwpilib/blob/master/wpilibjIntegrationTests/build.gradle#L22-L23).

The main method for running tests with Jenkins is the [AntJunitLanucher](https://github.com/wpilibsuite/allwpilib/blob/master/wpilibjIntegrationTests/src/main/java/edu/wpi/first/wpilibj/test/AntJunitLanucher.java) (which I also just realized that I misspelled (has that been that way for all these years?!?!)).

# A Better Path Forward

A much better way of doing this would be to use the new Junit 5 [Console Launcher](http://junit.org/junit5/docs/current/user-guide/#running-tests-console-launcher). The console launcher includes all of the features that I hacked into the original test framework. For example, class, package, or method selection.

As a first pass I would propose using the [junit-vintage-engine](http://junit.org/junit5/docs/current/user-guide/#migrating-from-junit4-running) and then porting tests to Junit 5 in the future.

Starlight220, 2020-09-14:
isn't this superseded by #1024?

AustinShalit, 2020-09-14:
No. This PR is talking about the integration tests. The other one is for the unit tests.

Starlight220, 2020-09-14:
Both involve updating to JUnit5, no? 
If so (and they should still be separate), then changing the title to something similar to "Update Integration Tests to use JUnit5" would be helpful.

calcmogul, 2021-12-31:
We're going to rewrite the tests for 2023 instead as per #327.

closed 2021-12-31 by calcmogul