wpilibsuite/allwpilib #638: SquaredInputs defaulting to true on TankDrive

Twinters007, 2017-09-15:
Seems like this shouldn't be the default. It creates unexpected outputs by default, which is confusing.

calcmogul, 2017-09-15:
What kind of unexpected outputs? Could you elaborate?

Twinters007, 2017-09-15:
Well the outputs to the motors will be squared, but I didn't specify squared outputs (sending a joystick value of .5 results in .25 going to motors)

calcmogul, 2017-09-15:
It's desirable to square the joystick inputs for open loop driving in teleop. Since that's a more common use case, it became the default.

We generally try to discourage teams from passing in raw values to RobotDrive functions for things like autonomous because feedback control approaches are more reproducible. I can understand teams not having a robot built for that (even though the KoP chassis includes encoders), but at that point they are using arbitrary trial-and-error values in RobotDrive for their autonomous anyway, so the scale doesn't really matter.

Twinters007, 2017-09-15:
What about teams that pass custom PID loop values into robotdrive? Then they have to worry about those values getting squared. 

I can understand why it would change due to high use, but it still may catch newer teams by surprise because it's a hidden feature. Is it more helpful to teams that want it than detrimental to teams that don't?

calcmogul, 2017-09-15:
If teams make a custom PIDOutput class or PID controller class that uses RobotDrive in some way, they should be advanced enough to read the documentation for WPILib classes they are using. If they use PID controllers for the left and right sides, a better approach in my opinion is to use the motor controllers directly for each side. Teams capable of implementing distance and angle pose controllers instead should be able to read the docs as well.

Teams should be reading the documentation for the APIs they use, or at the very least, looking at the header where the function is defined. If `squaredInputs` is as hidden as you say, then teams are either refusing to read the documentation or unaware the documentation exists. That seems like a bigger problem we need to address.

Twinters007, 2017-09-15:
I discovered this problem by being aware that tankdrive exists, along with some autocomplete, but didn't discover squaredInputs until someone pointed out that the reason my outputs were squared is that it was enabled by default. I didn't even think to look in the docs because I didn't think the problem lay in wpilib, but rather my own code. 

In regards to PID, I mean if you had a simple method that calculated the values to rotate (maybe to an angle with a gyro). Not complex enough to necessitate a PID controller, so you create a function that takes an angle, calculates error, multiplies by kP, and sends that to tankDrive. I guess since tuning PID is a guess and check, the squared inputs won't really make a difference....hmm

scarmain, 2017-09-15:
A suggestion I was just about to put in, but found this thread instead...

Could a method be added to RobotDrive to set squaring on/off (aka true/false)?  This local setting would be used in all the cases where squaredInputs is not provided as a parameter.  

In my experience, for auto mode, it is confusing to have squaring inputs (see discussion above), but teleop, it allows for much smoother control.  Having a switch to allow us to switch it on and off is a lot easier then changing a bunch of commands to add a false parameter (that no one understands why we added until they read the WpiLib code).

Twinters007, 2017-10-30:
Unit testing is also weird because if I don’t pass in squared the inputs become squared. Has any thought been given to this or is it not going to change for 2018?

calcmogul, 2017-10-30:
We discussed this internally and decided not to change it for 2018.

nickschatz, 2017-11-16:
I think it would reduce confusion to rename "squaredInputs" to "squareInputs" in the interest of self-documenting code. The current is ambiguous to whether the inputs to the function should already be squared or the inputs to the motors (the outputs of the function) should be squared. Changing this to "squareInputs" clearly describes that the method is performing an action based on whether this argument is true or false.

calcmogul, 2018-10-11:
Just filed #1360.

twinters007, 2018-10-12:
I still think it shouldn't default to true.

calcmogul, 2018-10-12:
The intended usage is in teleop, not autonomous. In teleop, having squaring on by default is desirable.

AustinShalit, 2019-08-04:
We are unable to make this change because it would be silently breaking.

closed 2019-08-04 by AustinShalit