wpilibsuite/allwpilib #642: LinearDigitalFilter in IIR/FIR mode returning nonzero value with uniformly zero input

Oblarg, 2017-09-18:
While debugging our drive code yesterday, we discovered that the LinearDigitalFilter class, in both IIR and FIR mode, can return (stable) nonzero values when given a uniformly zero input if the input has been nonzero at some point in the past.

We initially thought this may have been unique to the IIR filter, and that it was simply taking longer than we expected for the weights on the previous values to diminish sufficiently for the output to return to zero.  However, upon switching the IIR filter out for a simple moving average, the problem remained.

It seems likely that accumulated floating point errors are at fault, but I do not have the time to comb through the code right now to check where they might be coming from.

I highly suggest that the library be changed to use integer or fixed-point math for many of these functions, as floating point arithmetic is generally a bad idea for embedded systems for precisely this reason.

calcmogul, 2017-09-18:
I can't reproduce this problem with MovingAverage. Here's my test code: https://github.com/calcmogul/allwpilib/tree/linearfilter-test. If you add a print statement of the current filter value, you'll see the filter that uses the pulse data increases from 0 to 0.1666, then goes back down to zero after 6 data points.

calcmogul, 2017-11-20:
I wasn't able to reproduce this. Any new info @Oblarg?

Oblarg, 2017-11-20:
So, we're very confused; we re-wrote the code in question and the bug did not occur, and have not been able to isolate the cause.  Probably not a WPILib issue, given the inability to reproduce it.  It'd be very nice to finally figure out what's going on in the problematic code, though.

closed 2017-11-20 by Oblarg