wpilibsuite/allwpilib #649: Implement SPI and I2C simulation

pjreiniger, 2017-09-23:
In order to simulate things like the ADX* family of sensors, NavX, Pixy, etc, it would be nice to have have a method do simulate the read/write/transaction calls the library does.  I see two options

### Use existing callback scheme
* Create callbacks for read, write, and transaction
* Create uint8 buffers in the internal data class to store the simulation data
* Create setters that allow you modify the read buffer
I have prototyped this approach with my simulator, and it is a little bit gross, but it works.  The sequence diagram is complicated and bounces around a lot between the simulation implementation and the HAL_SIM classes.  There are almost certainly multi-threading concerns

### Create new callback types
* Force the simulation implementation to store the state of read/write/transaction
* Create a write callback that includes the buffer to write (and byte count)
* Create read callback that, with the assumption that the simulation implementation fills out the buffer
* Similar callback scheme for transaction, basically does both of the above.
With this approach you lose the ability to have multiple listeners, since they would be fighting over filling out the read buffer

@ThadHouse - you seem to be the simulator person.  I was wondering if you had any plans to do this simulation, or thoughts on my ideas before I start doing PRs

virtuald, 2017-09-24:
Using the existing callback scheme isn't that bad for most hardware types -- it's worth noting how RobotPy deals with this -- it's basically a mini-hal for each operation. It would be more difficult to implement well in Java/C++. I don't have time to write anything, but see the implementation at:

* https://github.com/robotpy/robotpy-wpilib/blob/master/wpilib/wpilib/adxrs450_gyro.py#L58
* https://github.com/robotpy/robotpy-wpilib/blob/master/hal-sim/hal_impl/spi_helpers.py

The NavX stuff is a bit more complex, but still it's not so bad:

* https://github.com/robotpy/robotpy-wpilib-utilities/blob/master/robotpy_ext/common_drivers/navx/registerio_i2c.py#L28
* https://github.com/robotpy/robotpy-wpilib-utilities/blob/master/robotpy_ext/common_drivers/navx/_impl/sim_io.py

pjreiniger, 2017-09-24:
In python you have the luxury of basically doing source overriding.  That is what my team did last year, but I'm trying to move 100% to the callback scheme, which is in this case is messy.

As I said, I have prototyped an implementation, but it is grosser than I think it should be for an open source solution

Simulators:
[I2C and SPI Accelerometers](https://github.com/pjreiniger/SnobotSim/tree/master/snobot_sim_java/java/snobot_sim_java/com/snobot/simulator/simulator_components/accelerometer)
[SPI Gyro](https://github.com/pjreiniger/SnobotSim/blob/master/snobot_sim_java/java/snobot_sim_java/com/snobot/simulator/simulator_components/gyro/ADXRS450_SpiGyroWrapper.java)
[I2C and SPI NavX](https://github.com/pjreiniger/SnobotSim/tree/master/snobot_sim_java/java/snobot_sim_java/com/snobot/simulator/simulator_components/navx)

Callback Scheme

[Attached](https://github.com/wpilibsuite/allwpilib/files/1327406/MockSpiI2c.zip)

pjreiniger, 2017-11-19:
Fixed int #688

closed 2017-11-19 by pjreiniger