wpilibsuite/allwpilib #661: Connecting to remote Gazebo sim

clebio, 2017-10-08:
Hello, I've posted this question to Chief Delphi, but not gotten any responses there. I realize this repo isn't for Gazebo, but figured you all would know as well as anyone. Basically, I'm trying to connect Eclipse (and sim_ds) to a remote Gazebo instance that's running on another computer on my LAN.

I've tried exporting `GAZEBO_MASTER_URI`:

    export GAZEBO_MASTER_URI=172.16.x.y:11345

But it seems that supplying the option at the command line, at runtime works better (at all?):

```
GAZEBO_MASTER_URI=172.16.x.y:11345 eclipse 
```

Same with `sim_ds`:

```
 $ GAZEBO_MASTER_URI=172.16.1.15:11345 ~/wpilib/simulation/sim_ds
INFO|net.java.games.input.DefaultControllerEnvironment: Loading: net.java.games.input.OSXEnvironmentPlugin
```

Before I was just getting `Connection refused` [1], but it appears right now that I get no errors. 

```
Buildfile: /Users/caleb/dev/frc/GearsBot/build.xml
Trying to override old definition of task classloader
compile:
     [echo] [athena-compile] Compiling src with classpath=/Users/caleb/wpilib/java/current/lib/WPILib.jar:/Users/caleb/wpilib/java/current/lib/NetworkTables.jar:/Users/caleb/wpilib/java/current/lib/opencv.jar:/Users/caleb/wpilib/java/current/lib/cscore.jar to build
jar-for-simulation:
     [echo] [jar-for-simulation] Building jar.
      [jar] Building jar: /Users/caleb/dev/frc/GearsBot/dist/FRCUserProgramSim.jar
simulate:
     [echo] [simulate] You may now run Gazebo and your DriverStation
     [echo] [simulate] Running Code.
     [java] resources = |sun.misc.CompoundEnumeration@1eb44e46|
     [java] platform: /Mac OS X/x86_64/
     [java] Failed to initialize simulator/dio/1/2: does the encoder exist?
     [java] NOTE|WPILibJSim: Encoder.setDistancePerPulse() assumes 360 pulses per revolution in simulation.
     [java] Failed to initialize simulator/dio/3/4: does the encoder exist?
     [java] NOTE|WPILibJSim: Encoder.setDistancePerPulse() assumes 360 pulses per revolution in simulation.
     [java] NOTE|WPILibJSim: Encoder.setDistancePerPulse() assumes 360 pulses per revolution in simulation.
     [java] NOTE|WPILibJSim: Encoder.setDistancePerPulse() assumes 360 pulses per revolution in simulation.
     [java] Default IterativeRobot.robotPeriodic() method... Overload me!
```

On the other hand, enabling teleop or autonomous doesn't do anything. When I run `sim_ds` and `eclipse` on the same machine where Gazebo is running, then both teleop and automous work fine.

[1]: https://www.chiefdelphi.com/forums/showthread.php?p=1702616#post1702616

clebio, 2017-10-08:
Well, I guess there is an error (though no `ERROR` in the line):

     Failed to initialize simulator/dio/1/2: does the encoder exist?
     Failed to initialize simulator/dio/3/4: does the encoder exist?

I see that on the working Eclipse (on the computer running Gazebo), that says instead

     Initialized simulator/dio/1/2
     Initialized simulator/dio/3/4

jwhite66, 2017-10-08:
That error message is generated by this code:
https://github.com/wpilibsuite/allwpilib/blob/v2017.3.1/wpilibc/sim/src/simulation/SimEncoder.cpp#L27

That implies that the Eclipse process hosting the simulator is unable to see the topic.

I'd be curious to see what you get with something like this:
  GAZEBO_MASTER_URI=172.16.x.y:11345 gz topic -l

clebio, 2017-10-08:
```
 $ gz topic -l
An instance of Gazebo is not running.
$ GAZEBO_MASTER_URI=172.16.1.15:11345 gz topic -l
Conflicting gazebo versions
/gazebo/default/Box/Box/wrench
/gazebo/default/GearsBot/backLeftWheel/wrench
/gazebo/default/GearsBot/backRightWheel/wrench
/gazebo/default/GearsBot/chassis/chassisUltrasonic/sonar
/gazebo/default/GearsBot/chassis/default/GearsBot/chassis/chassisUltrasonicsensor_collision/contacts
/gazebo/default/GearsBot/chassis/robotCamera/cmd
/gazebo/default/GearsBot/chassis/robotCamera/image
/gazebo/default/GearsBot/chassis/wrench
/gazebo/default/GearsBot/elevator/wrench
/gazebo/default/GearsBot/frontLeftWheel/wrench
/gazebo/default/GearsBot/frontRightWheel/wrench
/gazebo/default/GearsBot/joint_cmd
/gazebo/default/GearsBot/leftFinger/leftFingerContact
/gazebo/default/GearsBot/leftFinger/leftFingerContact/contacts
/gazebo/default/GearsBot/leftFinger/wrench
/gazebo/default/GearsBot/rightFinger/wrench
/gazebo/default/GearsBot/wrist/wrench
/gazebo/default/SodaCan/SodaCan/wrench
/gazebo/default/asphalt_plane/link/wrench
/gazebo/default/diagnostics
/gazebo/default/factory
/gazebo/default/ground_plane/link/wrench
/gazebo/default/gui
/gazebo/default/gzclient_camera/cmd
/gazebo/default/joint
/gazebo/default/light
/gazebo/default/log/control
/gazebo/default/log/status
/gazebo/default/model/info
/gazebo/default/model/modify
/gazebo/default/physics
/gazebo/default/physics/contacts
/gazebo/default/playback_control
/gazebo/default/pose/info
/gazebo/default/pose/local/info
/gazebo/default/pose/modify
/gazebo/default/request
/gazebo/default/response
/gazebo/default/roads
/gazebo/default/scene
/gazebo/default/selection
/gazebo/default/sensor
/gazebo/default/skeleton_pose/info
/gazebo/default/sky
/gazebo/default/user_camera/joy_pose
/gazebo/default/user_camera/joy_twist
/gazebo/default/user_camera/pose
/gazebo/default/visual
/gazebo/default/world_control
/gazebo/default/world_stats
/gazebo/frc/simulator/analog/1/control
/gazebo/frc/simulator/analog/1/position
/gazebo/frc/simulator/analog/1/velocity
/gazebo/frc/simulator/analog/2
/gazebo/frc/simulator/analog/3
/gazebo/frc/simulator/analog/6
/gazebo/frc/simulator/dio/1/2/control
/gazebo/frc/simulator/dio/1/2/position
/gazebo/frc/simulator/dio/1/2/velocity
/gazebo/frc/simulator/dio/3/4/control
/gazebo/frc/simulator/dio/3/4/position
/gazebo/frc/simulator/dio/3/4/velocity
/gazebo/frc/simulator/dio/5
/gazebo/frc/simulator/pwm/1
/gazebo/frc/simulator/pwm/2
/gazebo/frc/simulator/pwm/3
/gazebo/frc/simulator/pwm/4
/gazebo/frc/simulator/pwm/5
/gazebo/frc/simulator/pwm/6
/gazebo/frc/simulator/pwm/7
/gazebo/frc/time
/gazebo/server/control
/gazebo/world/modify
```

jwhite66, 2017-10-08:
Looking here:
  https://github.com/wpilibsuite/allwpilib/blob/v2017.3.1/simulation/JavaGazebo/src/main/java/org/gazebosim/transport/Connection.java#L78

It looks like our eclipse plugins use our own environment variable, rather than just using the gazebo uri.  It looks like it is GAZEBO_IP.  Doesn't look like it includes a port, but I didn't study the code closely.

clebio, 2018-01-28:
Any updates on this? I find that the 2018 documentation on screenstepslive no longer includes information on Gazebo sim at all. Worse yet, Google search results for "frc gazebo sim" come up showing "2017", but the links end up just being 2018 documents.

[image]
[image]

AustinShalit, 2019-08-04:
Duplicate of #456

closed 2019-08-04 by AustinShalit