wpilibsuite/allwpilib #663: HAL Simulator RestartTiming() call breaks PacGoat C++ Simulation PID Control

jwhite66, 2017-10-09:
I'm working to get the C++ PacGoat sample program fully functional with the Gazebo simulator.  The PID Control loop is presently failing.  I've tracked it down to this call:
  https://github.com/wpilibsuite/allwpilib/blob/master/hal/src/main/native/sim/HAL.cpp#L197

That resets the FPGA clock (effectively shrinks it).  However, the Robot has several PID control loops that have already been created at this point.  When the clock suddenly shifts downwards, the wait is essentially guaranteed to never complete.

I believe that it is the constructors for these global variables:
https://github.com/wpilibsuite/EclipsePlugins/blob/master/edu.wpi.first.wpilib.plugins.cpp/resources/templates/examples/PacGoat/src/Robot.cpp#L8
That are firing before the invocation to HAL_Initialize from /wpilibc/src/main/native/include/RobotBase.h.

I have confirmed that simply removing the call to RestartTiming() in HAL.cpp:
https://github.com/wpilibsuite/allwpilib/blob/master/hal/src/main/native/sim/HAL.cpp#L197

Does address this issue, but I do not know what other side effects it has.
That line went in as part of a very large patch:

commit e1195e8b9dab602f036da0c801ba0cc505730b80
Author: Thad House <ThadHouse@users.noreply.github.com>
Date:   Fri Aug 18 21:35:53 2017 -0700

    Update to 2018_v4 image and new build system. (#598)

So I don't have any visibility into why that line in particular is present.

Can we remove the call to RestartTiming()?

jwhite66, 2017-10-09:
Hmm.  Another piece that I should be working on is to make the clock being provided by Gazebo become the time reference source.  That implies that if I initialize the gazebo plugin at that same point, that the gazebo plugin will also want to cause a break in the clock.

Conceivably, we could have parts of the hal simulator trigger the HAL_Initialize; i.e. have HALSIM_GetFPGATime() invoke HALSIM_Initialize.

Arguably, the code should not be constructing objects before robot init (and if I change the code to forestall object creation, it does work).  But I fear that is subtlety that is going to be lost on students, and troubleshooting this was quite difficult.

While I'm writing full length novels, another thing I've discovered while working on this is that the smart dashboard components can have a similar problem.  If you have an object that calls a smart dashboard put before the robot init does a smartdashboard::init, you get a crash.  So perhaps we have two cases to consider on this one.

ThadHouse, 2017-10-09:
This is actually an issue we have seen before and don't actually have a way to fix. On a real robot, globals like that would fail as well, and there actually is no solution to that. I didn't realize the PacGoat example actually created the subsystems as globals, since the code it was originally written for was Java which doesn't have this issue. 

The reason this cannot be fixed is in the user cpp file, we cannot guarantee we can statically initialize a variable before anything else. And the only way to avoid this issue would be to put a HAL_Initialize call in EVERY constructor in wpilibc, which is not a feasible idea. 

In Java, this isn't an issue because globals are not allowed, but in C++ we just don't have a viable way of fixing this issue. Also, that RestartTiming() call is needed because otherwise all the time variables are not initialized.

For now, the real fix for that is going to be move all of the construction of those pointers into the constructor of the main class. I would actually consider that an example bug and we should fix that way.

As for switching clocks, I don't actually think that is necessary, and might not match the actual roborio behavior. The clock on the rio is actually not a time clock, but rather a tick clock counting time from boot, and that never changes. It also has microsecond timing. In the sim, we were able to get 10us timing (I think) and I highly doubt Gazebo's clock has both of these features.

AustinShalit, 2019-08-04:
https://github.com/wpilibsuite/allwpilib/commit/434d60592cece628cc068f0be9bcaaf42818278e Is the best fix for this we can provide at the moment.

closed 2019-08-04 by AustinShalit