wpilibsuite/allwpilib #683: Compatibility issue with talon libraries

romi2002, 2017-10-23:
When wpilib is included before the talon libraries, it returns the folowing error: `error: 'TALON_Control_7_FeedbackOutputOverride_t' was not declared in this scope
   CtreCanNode::txTask<TALON_Control_7_FeedbackOutputOverride_t> GetControl7();`. This can be fixed by adding the missing struct: `TALON_Control_7_FeedbackOutputOverride_t` in `ctre_frames.h`.

romi2002, 2017-10-23:
The missing struct being:
 ```
typedef struct _TALON_Control_7_FeedbackOutputOverride_t {
	unsigned OnTime3Ms:8;
	unsigned OnTime4Ms:8;
	unsigned OnTime5Ms:8;
	unsigned OnTime6Ms:8;
	unsigned OnTime7Ms:8;
	unsigned OnTime8Ms:8;
	unsigned PeriodMs:8;
	unsigned reserved0:2;
	unsigned OutputEnable_Ch3:1;
	unsigned OutputEnable_Ch4:1;
	unsigned OutputEnable_Ch5:1;
	unsigned OutputEnable_Ch6:1;
	unsigned OutputEnable_Ch7:1;
	unsigned OutputEnable_Ch8:1;
} TALON_Control_7_FeedbackOutputOverride_t;
```
 (from the talon libs)

JacisNonsense, 2017-10-24:
This issue seems to affect all cases where `ctre_frames.h` is included before `ctre/structs_mtrCntrl.h`, mostly due to them sharing an include guard (`CTRE_FRAMES_H`). An example of this is including the PCM header files (`PCM.h` -> `ctre.h` -> `ctre_frames.h`).

I feel this should be exclusively the job of the CTRE Toolsuite to define these structures, not WPILib. I understand the choice of moving CTRE products to their own library, but doing it half and half (i.e. Talon SRX but not PCM, both of which are used very heavily) leads to issues like this. Keeping CTRE's devices segmented between implementation in the Toolsuite and implementation in WPILib could lead to a ton more issues like this, and requires that the header files be watched diligently.

@ozrien I dare say this is much more up your alley.

PR #728 (2017-11-14): Move ctre headers to hal/src so they aren't user-visible.
Also removes ctre_frames.h entirely since it's not used.

Fixes #683.
commit 020ee227 (2017-11-14): Move ctre headers to hal/src so they aren't user-visible. (#728)

Also removes ctre_frames.h entirely since it's not used.

Fixes #683.
closed 2017-11-14 by PeterJohnson